mavlink/MAVSDK-Java #13: Mission speed seems to be ignored

RyanHurst, 2019-04-15:
Missions seems to be run at the same speed regardless of what speedMS is passed in to the MissionItem.

julianoes, 2019-04-17:
I think this is a recent PX4 Firmware bug presumably fixed now:
https://github.com/PX4/Firmware/issues/11764

JonasVautherin, 2019-04-17:
@RyanHurst which version of the Firmware are you using? Could you confirm that?

JonasVautherin, 2019-04-23:
@RyanHurst Did you have a chance to check the Firmware version?

RyanHurst, 2019-04-25:
My firmware is built from commit e6f1a2db12def90cbe849ffe8898b5547b72e5f5 (april 2nd)

julianoes, 2019-04-26:
You need something after April 15.
https://github.com/PX4/Firmware/pull/11849
https://github.com/PX4/Firmware/pull/11836

julianoes, 2019-06-06:
Closing this, please re-open if its a problem again.

closed 2019-06-06 by julianoes