mavlink/MAVSDK-Java #14: Uploading a mission will crash in Android api level <24

RyanHurst, 2019-04-16:
Steps to reproduce: 
Open Mission sample app on an android device running lollipop.
Upload a mission.

If you open up Mission.java, Android Studio provides a warning on line 318:
`Method reference requires API level 24 (current min is 21): missionItems.stream().map(elem -> elem.rpcMissionItem())::iterator`

Logcats:
```
04-16 11:27:11.988 6063-6063/io.dronecore.dronecoreclient E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.dronecore.dronecoreclient, PID: 6063
    java.lang.NoSuchMethodError: No interface method stream()Ljava/util/stream/Stream; in class Ljava/util/List; or its super classes (declaration of 'java.util.List' appears in /system/framework/core-libart.jar)
        at io.dronecode_sdk.mission.Mission.uploadMission(Mission.java:318)
        at io.dronecore.dronecoreclient.MapsViewModel.startMission(MapsViewModel.java:58)
        at io.dronecore.dronecoreclient.MapsActivity.lambda$onCreate$0(MapsActivity.java:67)
        at io.dronecore.dronecoreclient.-$$Lambda$MapsActivity$XPnl_5WvEWE9lr8HVMHD3eIKKLc.onClick(lambda)
        at android.view.View.performClick(View.java:4780)
        at android.view.View$PerformClick.run(View.java:19866)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:135)
        at android.app.ActivityThread.main(ActivityThread.java:5254)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
```

JonasVautherin, 2019-04-23:
Too bad, that syntax was elegant :-). Let's move back to something backward compatible then. 

@RyanHurst would you have a suggestion on how to write that for older versions of Java? That would be for [here](https://github.com/dronecore/DroneCore-Java/blob/master/sdk/templates/request.j2#L8) and [here](https://github.com/dronecore/DroneCore-Java/blob/master/sdk/templates/call.j2#L8).

RyanHurst, 2019-04-25:
```
        List<MissionProto.MissionItem> protoMissionItems = new ArrayList<>();
        for (Mission.MissionItem missionItem : missionItems) {
            protoMissionItems.add(missionItem.rpcMissionItem());
        }

        MissionProto.UploadMissionRequest request = MissionProto.UploadMissionRequest
                .newBuilder()
                .addAllMissionItems(protoMissionItems)
                .build();
```
I'm not sure what our other options would be.

JonasVautherin, 2019-04-25:
Your solution makes sense, but it's definitely more verbose :sweat_smile:.

I am actually wondering if, instead of using a java `stream()` we could use RxJava and go through an `Observable`. That could work just as well as the original syntax.

JonasVautherin, 2019-04-26:
Thinking about it, I realize that [using RxJava](https://github.com/ReactiveX/RxJava/wiki/Creating-Observables#fromiterable-example) would probably be the right way to go:

Something of the kind:

```java
Observable<MissionItem> observable = Observable.fromIterable(list).map(...);
```

And the SDK should probably not expose a `List`, but always an `Observable` (or `Flowable`, that's to be verified). I think it is more consistent in RxJava. And if users of the SDK then want a list, they can always get it from the `Observable` (i.e. a `List` is the "blocking" version of an `Observable`).

closed 2022-09-23 by JonasVautherin