mavlink/MAVSDK-Java #15: Usage Examples

anandbaskaran, 2019-06-17:
Firstly I thank the authors for the amazing work. The examples provided with the repo clearly defines the working of the client applications. However, having examples (and/or) working of drone connectivity, autodiscovery, the architecture of the project would be very helpful.

Thanks in advance

JonasVautherin, 2019-06-17:
Hello!

That's a good point, and we will add new examples as we go (for instance you can see that we have [more examples in Python](https://github.com/mavlink/MAVSDK-Python/tree/master/examples) and we have a [Swift sample app for iOS](https://github.com/mavlink/mavsdk-swift-example)). Because RxSwift and RxJava are quite similar, I expect those examples can help you (until we add the java ones).

In the meantime, are you aware that the full API is define [from proto files here](https://github.com/mavlink/MAVSDK-Proto/tree/master/protos)? That should help you find the features we support.

Feel free to ask questions if something is unclear, and if you play with the SDK, please feel free to add examples! We will be more than happy to review your PRs :-).

JonasVautherin, 2019-07-18:
Closing at it is quite a general question. Feel free to open new issues if you have questions.

If you managed to run MAVSDK-Java, feel free to add new examples!

closed 2019-07-18 by JonasVautherin