mavlink/MAVSDK-Java #17: SDK build faild

damianledzinski, 2019-07-09:
Hello,
I can not build sdk. When I try to build sdk, the result is:
```
damian@jarvis:~/git/MAVSDK-Java/sdk$ ./gradlew build 

> Task :dronecode-sdk:generateProto FAILED
protoc plugin 'dcsdk' not defined. Trying to use 'protoc-gen-dcsdk' from system path

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':dronecode-sdk:generateProto'.
> protoc: stdout: . stderr: protoc-gen-dcsdk: program not found or is not executable
  --dcsdk_out: protoc-gen-dcsdk: Plugin failed with status code 1.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 0s
3 actionable tasks: 1 executed, 2 up-to-date
```
And running with --info option, gives:
```
> Task :dronecode-sdk:extractIncludeProto UP-TO-DATE
Skipping task ':dronecode-sdk:extractIncludeProto' as it is up-to-date.
:dronecode-sdk:extractIncludeProto (Thread[Execution worker for ':',5,main]) completed. Took 0.02 secs.
:dronecode-sdk:extractProto (Thread[Execution worker for ':',5,main]) started.

> Task :dronecode-sdk:extractProto UP-TO-DATE
Skipping task ':dronecode-sdk:extractProto' as it is up-to-date.
:dronecode-sdk:extractProto (Thread[Execution worker for ':',5,main]) completed. Took 0.001 secs.
:dronecode-sdk:generateProto (Thread[Execution worker for ':',5,main]) started.

> Task :dronecode-sdk:generateProto FAILED
Task ':dronecode-sdk:generateProto' is not up-to-date because:
  Task has failed previously.
Custom actions are attached to task ':dronecode-sdk:generateProto'.
Resolving artifact: [group:com.google.protobuf, name:protoc-gen-javalite, version:3.0.0, classifier:linux-x86_64, ext:exe]
Resolved artifact: /home/damian/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protoc-gen-javalite/3.0.0/ff5b99fa1e45b51e0350b8da0fa20d47677b3a09/protoc-gen-javalite-3.0.0-linux-x86_64.exe
Resolving artifact: [group:io.grpc, name:protoc-gen-grpc-java, version:1.19.0, classifier:linux-x86_64, ext:exe]
Resolved artifact: /home/damian/.gradle/caches/modules-2/files-2.1/io.grpc/protoc-gen-grpc-java/1.19.0/f833e69df003f88d975fcb55a464e6e2d1954eb4/protoc-gen-grpc-java-1.19.0-linux-x86_64.exe
Resolving artifact: [group:com.google.protobuf, name:protoc, version:3.6.1, classifier:linux-x86_64, ext:exe]
Resolved artifact: /home/damian/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protoc/3.6.1/f8584d1e86d1adb61d9c559e4b49cedbc97c4d78/protoc-3.6.1-linux-x86_64.exe
file or directory '/home/damian/git/MAVSDK-Java/sdk/dronecode-sdk/src/main/proto', not found
protoc plugin 'dcsdk' not defined. Trying to use 'protoc-gen-dcsdk' from system path
[/home/damian/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protoc/3.6.1/f8584d1e86d1adb61d9c559e4b49cedbc97c4d78/protoc-3.6.1-linux-x86_64.exe, -I/home/damian/git/MAVSDK-Java/sdk/proto/protos, -I/home/damian/git/MAVSDK-Java/sdk/dronecode-sdk/build/extracted-protos/main, -I/home/damian/git/MAVSDK-Java/sdk/dronecode-sdk/build/extracted-include-protos/main, --dcsdk_out=file_ext=java,template_path=/home/damian/git/MAVSDK-Java/sdk/templates/:/home/damian/git/MAVSDK-Java/sdk/dronecode-sdk/build/generated/source/proto/main/dcsdk, --plugin=protoc-gen-grpc=/home/damian/.gradle/caches/modules-2/files-2.1/io.grpc/protoc-gen-grpc-java/1.19.0/f833e69df003f88d975fcb55a464e6e2d1954eb4/protoc-gen-grpc-java-1.19.0-linux-x86_64.exe, --grpc_out=lite:/home/damian/git/MAVSDK-Java/sdk/dronecode-sdk/build/generated/source/proto/main/grpc, --plugin=protoc-gen-javalite=/home/damian/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protoc-gen-javalite/3.0.0/ff5b99fa1e45b51e0350b8da0fa20d47677b3a09/protoc-gen-javalite-3.0.0-linux-x86_64.exe, --javalite_out=/home/damian/git/MAVSDK-Java/sdk/dronecode-sdk/build/generated/source/proto/main/javalite, /home/damian/git/MAVSDK-Java/sdk/proto/protos/action/action.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/calibration/calibration.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/camera/camera.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/core/core.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/gimbal/gimbal.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/info/info.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/mission/mission.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/param/param.proto, /home/damian/git/MAVSDK-Java/sdk/proto/protos/telemetry/telemetry.proto]
:dronecode-sdk:generateProto (Thread[Execution worker for ':',5,main]) completed. Took 0.047 secs.

FAILURE: Build failed with an exception.
```
How to solve it?

Thanks in advance

JonasVautherin, 2019-07-10:
Hello!

I am updating the documentation and the way it is built, I'll keep you posted on this issue.

The error here is that it does not find `protoc-gen-dcsdk`, which is our protoc plugin. You would need to build it yourself (README in `proto/pb_plugins`) and add it to your PATH. I will version the generated files such that you don't need to do that in the future :+1:. That should be updated in the next few days.

damianledzinski, 2019-07-10:
Thank you for your quick response.
It works :)

JonasVautherin, 2019-07-10:
Great! Happy to read that :blush:!

closed 2019-07-10 by JonasVautherin

ashirsat, 2020-07-23:
@JonasVautherin I am still unable to find '''protoc-gen-dcsdk''' in my path. Can you elaborate how to add it to the path?

JonasVautherin, 2020-07-24:
@ashirsat: What I meant above is that you need to install it from sources, as documented [here](https://github.com/mavlink/MAVSDK-Proto/tree/471aa4aa701bf5589a70ca65470ee7def96c570c/pb_plugins):

From `proto/pb_plugins`:

```
pip install -r requirements.txt
pip install -e .
```

Then you have to make sure it is in your path (I typically do the installation step (as described above) in a venv, and it's then automatically in the PATH).

Note that MAVSDK-Java still uses `protoc-gen-dcsdk` that you need to install from sources. There is now `protoc-gen-mavsdk` (which is the same, but renamed) available on pypi with `pip install protoc-gen-mavsdk`. But MAVSDK-Java [is not using it yet](https://github.com/mavlink/MAVSDK-Java/blob/master/sdk/build.gradle#L67). It may just be a matter of replacing `dcsdk` with `mavsdk` [here](https://github.com/mavlink/MAVSDK-Java/blob/master/sdk/build.gradle#L67), but I haven't tried it.

I would be glad if you could contribute that update :blush:.

elinxer, 2021-10-12:
i am still this error:
```
protoc plugin 'mavsdk' not defined. Trying to use 'protoc-gen-mavsdk' from system path
```
on the win10,py3,and add to system path sure:
```
C:\Users\elinx\AppData\Local\Programs\Python\Python310\Lib\site-packages\protoc_gen_mavsdk
```
i need help...

JonasVautherin, 2021-10-12:
That's a different error. Can you open a new issue with it? Also can you copy-paste the error message instead of posting a screenshot? That's easier to work with.

lxzua, 2023-12-26:
when I build in project ……\MAVSDK-Java\mavsdk_server, mavsdk_server_android-x86_v1.4.16.tar is empty.  And no extractForArch("x86-64").
A problem occurred evaluating root project 'mavsdk-server'.

[image]

What should I do fix it?
Thanks a lot!