mavlink/MAVSDK-Java #20: Missing classes

Ayman-git, 2019-07-28:
I have some classes that are missing and that classes are Action and Mission and Telemetry Classes.

JonasVautherin, 2019-07-28:
Can you elaborate on that? What are you doing, on what platform, and what is the error message? Do you have a small example that I could reproduce?

Ayman-git, 2019-07-28:
I am using android studio, 
[image]
[image]

JonasVautherin, 2019-07-28:
Can you try the Java example, taking care of removing the line I linked above?

And in your Android project, can you remove the same line in settings.gradle and try again? Make sure you clean build.

Ayman-git, 2019-07-28:
I re imported the whole file and the errors are gone, but i cant run the program
[image]

Ayman-git, 2019-07-28:
Found error in activity main this time
[image]

JonasVautherin, 2019-07-28:
Would you mind trying to do what I asked you to try above?

Ayman-git, 2019-07-28:
Yeah I removed them after re-importing

JonasVautherin, 2019-07-28:
What do you mean by re-importing? You're using gradle, right?

Did you try the java example? What's the output of `./gradlew build` in `java-client`, after you have removed the line in `settings.gradle`?

Ayman-git, 2019-07-29:
Nope, I am trying to run the app, but that isn't working after deleting it and re-importing it in another way and re-importing it removed the previous errors but now I have errors in activity main file.

Ayman-git, 2019-07-29:
Can you send me your new project that is working and i will try importing it.

JonasVautherin, 2019-07-29:
I am sorry, but I can't help much more if you don't give me more than "it doesn't work". I would need error messages (not screenshots showing that it doesn't work), and/or I would need you to try the steps I give you. The last thing I can ask now is whether you actually edited [this line](https://github.com/mavlink/MAVSDK-Java/blob/master/examples/android-client/settings.gradle#L3) (i.e. in `android-client`), since you run `android-client` and not `java-client`.

I know that `java-client` is working for me right now, which is why I would like to see if that works for you as well. Also, [MAVSDK-Java](https://search.maven.org/search?q=a:mavsdk) is now available on Maven Central, which is why I asked you to edit `settings.gradle`. If you don't, gradle will try to build the SDK from sources and that will fail if you don't set it up correctly (i.e. you would need `protoc-gen-dcsdk` in your path).

If you don't want to try `java-client`, I can try the Android sample again to confirm that it works for me. But I won't have time to do that before another few days.

Maybe @RyanHurst could tell us whether `android-client` works for him right now? Remember to try without [this line](https://github.com/mavlink/MAVSDK-Java/blob/master/examples/android-client/settings.gradle#L3), so that it gets the package from Maven Central! :blush:

Ayman-git, 2019-07-29:
This is the error i got, i opened "java-client" to check if it works for me and this is what happened, and i removed the line you asked me to remove
[image]

JonasVautherin, 2019-07-29:
It complains that the included build does not exist. Which seems to suggest that `includeBuild` is still setup somewhere. What happens if you clean build? Can you show me a screenshot of this project with `settings.gradle` opened, so that I confirm that the line got removed?

Ayman-git, 2019-07-30:
[image]

JonasVautherin, 2019-07-30:
Somehow it still believes that `includeBuild` is set. Can you try to clean build, or clean cache, or invalidate cache, or import again?

Ayman-git, 2019-07-30:
So I tried everything you told me, after seeing no change I imported it again but still no change and this is what i got.
[image]

Ayman-git, 2019-07-30:
I am facing same problem for both " android-client" and "java-client"

JonasVautherin, 2019-07-30:
The error reads: `ERROR: Included build 'C:\Users\Ayman\sdk' does not exist.`.

This path seems to be `../../sdk` from your imported project. And `../../sdk` is specified in `settings.gradle`: `includeBuild '../../sdk'`.

I don't see another place where that is defined, and you say that you removed that line, cleaned the cache and also started the project from scratch.

I'm not sure what else I can do here. If I were you, I would try the java example in command line (`./gradlew build`), and I would also try to clone the repo again in a new place, and try the java client there. If your Android-Studio still looks for `../../sdk` after you have removed that line from `settings.gradle`, it means that it is still saved somewhere, somehow.

Ayman-git, 2019-07-31:
Thanks for your help I will try that if didn't work then I will just try another code.

closed 2019-07-31 by Ayman-git

JonasVautherin, 2019-07-31:
Sure. Let me know if you get a different error when trying in a fresh git clone. That may bring more information.