mavlink/MAVSDK-Java #25: Unable to build mavsdk_server for armv7 android device

gparrow, 2019-12-09:
Hello,

I'm having difficulty building the backend (C++) for armeabi-v7a.   Following the instructions for building from source do not seem to work (get a message about the system no longer using makefiles, instructions coming later?) but I've made what seems to be progress by using the following below, strung together from a variety of web sources for each issue that comes up.   It seems the biggest problem is getting gRPC, error below: 

> CMake Error at cmake/cares.cmake:38 (find_package):
>   Could not find a package configuration file provided by "c-ares" with any
>   of the following names:
> 
>     c-aresConfig.cmake
>     c-ares-config.cmake
> 
>   Add the installation prefix of "c-ares" to CMAKE_PREFIX_PATH or set
>   "c-ares_DIR" to a directory containing one of the above files.  If "c-ares"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> Call Stack (most recent call first):
>   CMakeLists.txt:135 (include)

command I've attempted  (hopefully you can suggest something simpler??) that seems to get somewhere: 
`cmake -G"Unix Makefiles" -DCMAKE_SYSTEM_NAME="Android" -DCMAKE_INSTALL_PREFIX=install -DCMAKE_SYSTEM_VERSION=21 -DCMAKE_ANDROID_NDK=${HOME}/Android/android-ndk-r18b -DCMAKE_TOOLCHAIN_FILE=${HOME}/Android/android-ndk-r18b/build/cmake/android.toolchain.cmake -DCMAKE_ANDROID_ARCH_ABI="armeabi-v7a" -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DBUILD_BACKEND=ON -DgRPC_INSTALL=ON -DgRPC_BUILD_TESTS=OFF -DgRPC_PROTOBUF_PROVIDER=package -DgRPC_ZLIB_PROVIDER=package -DgRPC_CARES_PROVIDER=package -DgRPC_SSL_PROVIDER=package -DRUN_HAVE_STD_REGEX=0 -DRUN_HAVE_POSIX_REGEX=0  -DRUN_HAVE_STEADY_CLOCK=0 -Bbuild/android -H.
`

Please let me know if I'm doing anything wrong?  I've attempted this on both OSX (10.14.6) and Ubuntu 18.04

I'm looking to build mavsdk_server to install on an armv7 android device. I attempted to use a few different NDK versions as well, 20b, 18b, 17c.

julianoes, 2019-12-09:
We suggest to use [dockcross](https://github.com/dockcross/dockcross) as described here:
https://mavsdk.mavlink.io/develop/en/contributing/build.html#cross_compilation_dockcross

gparrow, 2019-12-09:
ok, so using dockcross at least builds something, but deploying to the device I get
`./mavsdk_server": error: Android 5.0 and later only support position-independent executables (-fPIE).`

I've tried passing the desired flags (and setting android-platform=21) and rebuilding but have not been able to get something built on either android-arm or android-arm64, but its quite possible I'm not passing things quite right - e.g. below

```
./dockcross-android-arm64 cmake -DANDROID_PLATFORM=android-21  -DCMAKE_BUILD_TYPE=Release -DBUILD_BACKEND=ON -DBUILD_SHARED_LIBS=OFF -DCMAKE_POSITION_INDEPENDENT_CODE=ON -Bbuild/android-arm64 -S.

./dockcross-android-arm64 cmake  --build build/android-arm64 -j 8
```

 Any further suggestions?

julianoes, 2019-12-10:
@JonasVautherin do you have an idea here?

gparrow, 2019-12-10:
Success!  I added the below lines to the root CMakeLists.txt and now have mavsdk_server running on the device!

```
# Use "-fPIC" / "-fPIE" for all targets by default, including static libs
set(CMAKE_POSITION_INDEPENDENT_CODE ON)

# CMake doesn't add "-pie" by default for executables (CMake issue #14983)
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -pie")
```

Once above added, used the command 
```
./dockcross-android-arm64 cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_BACKEND=ON -DBUILD_SHARED_LIBS=OFF  -Bbuild/android-arm64 -S.
  ./dockcross-android-arm64 cmake  --build build/android-arm64 -j 8
```

JonasVautherin, 2019-12-10:
Great!

BTW, I see that the issue mentions armv7, and you use arm64 in the solution. There is an arm64-v8a version [released on mavenCentral](https://mvnrepository.com/artifact/io.mavsdk/mavsdk-server) already, and used [in the android example](https://github.com/mavlink/MAVSDK-Java/blob/master/examples/android-client/app/build.gradle#L63).

closed 2019-12-10 by JonasVautherin

gparrow, 2019-12-10:
OK Thanks! can confirm the android-arm dockcross builds (and runs) for armv7 too, for anyone who comes looking