mavlink/MAVSDK-Java #26: Is there a possibility to launch MAVSDK Java server on Ubuntu?

KrisRadowski, 2019-12-13:
Hey,

I want to launch MAVSDK Java server on Ubuntu, because I'm simulating there drones. Is it possible? (If yes, how to do that?)

JonasVautherin, 2019-12-14:
Hello!

I am not sure I understand exactly your concern. If what you want to do is write a Java application using MAVSDK, and run it on your Ubuntu machine to control drones, then yes, that's possible. Did you have a look at the [java examples](https://github.com/mavlink/MAVSDK-Java/tree/master/examples/java-client)?

Note that the drones don't have to be simulated on your machine. All you need is a MAVLink connection between your application and the drone. Some examples:

* Run MAVSDK-Java on an Android phone, and control a real drone
* Run MAVSDK-Java on an Android phone, and control a drone simulated on an Ubuntu machine
* Run MAVSDK-Java on an Ubuntu machine, and control a drone simulated on an Ubuntu machine (possibly the same machine)
* Run MAVSDK-Java on an Ubuntu machine, and control a real drone

Does that help?

KrisRadowski, 2019-12-14:
yes, that's does it, thanks for help :)

closed 2019-12-14 by KrisRadowski