mavlink/MAVSDK-Java #28: No System error in android-client

SongeunYoo, 2020-01-28:
Hello. I have question in android-client example.
I followed java-client README.md (cause of no README file in android-client), but there're some problems in running android-client example.
When running it, the error occurred
"W/System.err: io.reactivex.exceptions.OnErrorNotImplementedException: NO_SYSTEM: No system"
[image]

I think it's because there's no simulator…… right?
The java README file says run a simulator(I used jMAVSim) mavsdk_server, after running a server. But I think it doesn't work for android example.

Is there any way to run simulator for android or run the example successfully?
Thanks.

JonasVautherin, 2020-01-28:
Hello! Did you manage to run the java-client example?

On Android, the sample app should "just work", but you may need to point the simulator to it. I usually do run a headless simulator using [this](https://github.com/JonasVautherin/px4-gazebo-headless#run-with-a-custom-ip-for-the-second-mavlink-interface), as it helps me sending mavlink to my phone.

So for instance, say that the android phone has the IP `192.168.0.12`, I run (from my laptop on the same network as the android phone):

```
$ docker run --rm -it jonasvautherin/px4-gazebo-headless:v1.9.2 192.168.0.12
```

SongeunYoo, 2020-01-29:
Oh, thx :D I'll try it.

closed 2020-01-29 by SongeunYoo

lbb1993, 2020-06-23:
> Hello! Did you manage to run the java-client example?
>
> On Android, the sample app should "just work", but you may need to point the simulator to it. I usually do run a headless simulator using [...]

Hi,I met the same problem.
if i use the jmavsim ,how could i solve this problem?

JonasVautherin, 2020-06-23:
@lbb1993: Does it work with the docker command you pasted above? If yes, one way with jMAVsim would be to use [mavlink-router](http://github.com/intel/mavlink-router) to send the jMAVsim packets to your Android device.