mavlink/MAVSDK-Java #29: java.net.ConnectException: Connection refused (Connection refused) error when running "./gradlew setRtlAltitude"

enymuss, 2020-02-23:
Hello,

Looking for help to fix this issue, to find the bug in how the java-client has been set up on my local machine.
Going through the java-client example, mavsdk_server and headless gazebo runs, but `./gradlew setRtlAltitude` comes up with an error.

MAVSDK server runs and connects to Gazebo.
```
MartaS:MAVSDK userX$ build/release/install/bin/mavsdk_server
[05:03:24|Info ] MAVSDK version: 0.24.0 (mavsdk_impl.cpp:25)
[05:03:24|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:401)
[05:03:24|Info ] Server started (grpc_server.cpp:38)
[05:03:24|Info ] Server set to listen on 0.0.0.0:51959 (grpc_server.cpp:39)
[05:03:24|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22)
[05:03:47|Info ] New system on: 127.0.0.1:60847 (udp_connection.cpp:264)
[05:03:47|Debug] Component Autopilot (1) added. (system_impl.cpp:386)
[05:03:47|Debug] MAVLink: info: [logger] file: ./log/2020-02-23/16_03_47.ulg (system_impl.cpp:292)
[05:03:48|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:556)
[05:03:48|Info ] System discovered [UUID: 5283920058631409231] (connection_initiator.h:61)
```

Running headless gazebo:
`docker run --rm -it jonasvautherin/px4-gazebo-headless:v1.10.1`
```
INFO  [simulator] Simulator connected on TCP port 4560.
[Msg] Using MAVLink protocol v2.0
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2020-02-23/16_03_47.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
  MAV_BROADCAST: curr: 0 -> new: 1
INFO  [px4] Startup script returned successfully
pxh> INFO  [ekf2] Mag sensor ID changed to 197388
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] 728000: EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] 5196000: EKF commencing GPS fusion
```

QGroundControl connects and I am able to Take Off and Land manually through QGroundControl. 
INFO output is printed for PX4 and MAVSDK server, showing take off and landing took place.

`./gradlew takeoffAndLand` builds successfully, but maybe it is a silent fail, as nothing shows up in QGroundControl controller, mavsdk_server or simulator logs.

```
./gradlew takeoffAndLand
> Task :takeoffAndLand
17:21:50.199 [main] DEBUG io.mavsdk.example.TakeoffAndLand - Starting example: takeoff and land...

BUILD SUCCESSFUL in 1s
2 actionable tasks: 1 executed, 1 up-to-date
```

Output with error from running `./gradlew setRtlAltitude`

```
MartaS:java-client userX$ ./gradlew setRtlAltitude

> Configure project :

=======
WARNING: this is not the best way to run `jshell`, as it may not be cross-platform and it won't have auto-completion enabled.
You may want to run the following instead:

    $ ./gradlew exportDeps
    $ jshell --class-path $(ls -d build/jshell_classpath_export/* | tr '\n' ':')
=======

[... 78 lines cut ...]
Caused by: java.net.ConnectException: Connection refused (Connection refused)
        at java.net.PlainSocketImpl.socketConnect(Native Method)
        at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
        at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
        at java.net.Socket.connect(Socket.java:607)
        at java.net.Socket.connect(Socket.java:556)
        at java.net.Socket.<init>(Socket.java:452)
        at java.net.Socket.<init>(Socket.java:262)
        at io.grpc.okhttp.OkHttpClientTransport$4.run(OkHttpClientTransport.java:509)
        ... 4 more
```

This is a clean build with the most recent git commits. 
Mavsdk_server was built from the master branch, with the same settings as outlined for building for macOS release in GitHub workflow [main.yaml file](https://github.com/mavlink/MAVSDK/blob/20875db04c6c6338d463cae58bfe6768edaf3e87/.github/workflows/main.yml).

Java version:
```
java -version
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_242-b08)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.242-b08, mixed mode)
```
Proto version: `proto (b34da69)`

JonasVautherin, 2020-02-23:
The example runs [`System drone = new System();`](https://github.com/mavlink/MAVSDK-Java/blob/master/examples/java-client/src/main/java/io/mavsdk/example/SetRtlAltitude.java#L14), which will [by default](https://github.com/mavlink/MAVSDK-Java/blob/master/sdk/mavsdk/src/main/java/io/mavsdk/System.java#L32) connect to `mavsdk_server` on port 50051.

So you should run `mavsdk_server -p 50051` instead. Without the `-p <port>` argument, `mavsdk_server` starts on a random port, in your logs above it's 51959.

```
[05:03:24|Info ] Server set to listen on 0.0.0.0:51959 (grpc_server.cpp:39)
```

If that helps, would you mind contributing to the documentation to make that clearer? Just opening a PR on the README.md would be super helpful already!

JonasVautherin, 2020-02-27:
@enymuss: Thanks a lot for the PR! Can we close this issue?

enymuss, 2020-02-27:
@JonasVautherin Yes, thank you for the help.

closed 2020-02-27 by enymuss