mavlink/MAVSDK-Java #31: [android-client] [grpc-default-executor-4] - Failed to upload the mission

enymuss, 2020-02-29:
When trying to run the android-client example, the app opens, the mapView loads and pressing on the screen adds waypoints. 
After adding the waypints, pressing the fab does nothing visable on screen. Logcat prints 
`io.mavsdk.androidclient E/MapsViewModel: [grpc-default-executor-4] - Failed to upload the mission` 
Pressing Land, Return, Kill or Takeoff all kill the app and it closes.

In the uploadMission, translateFromRpc() returns result.result as ERROR and result.resultStr as also Error.

It looks like it is sending something to a ?simulator?, but can't, so I've tried changing the ip addresses, looking for it to connect.

#Things Tried
- running on a virtual device in android studio, but can't build due to lacking native.so library, so moved to on-device android testing
- Changing the BACKEND_IP_ADDRESS to a) phone IP and b) to gazebo-headless broadcasting ip `network interface eth0, IP: 172.17.0.2`. 
This makes the warning disappear, but nothing happens on screen.
- running `docker run --rm -it jonasvautherin/px4-gazebo-headless:v1.10.1 PHONE_IP`, but the docker logs never show something connected to it and the error appears.

Running this on Android version 9.

What can I try to get the android-client example working?

julianoes, 2020-03-01:
Thanks for the issue, hopeully @JonasVautherin can look into this soon.

JonasVautherin, 2020-03-04:
Does it work for other examples, like taking off?

`BACKEND_IP_ADDRESS` is the IP where `mavsdk_server` is running. On Android that would be `127.0.0.1`, unless you run `mavsdk_server` on your computer (which I sometimes do for debugging).

`docker run --rm -it jonasvautherin/px4-gazebo-headless:v1.10.1 PHONE_IP` should do it, if the computer running that can reach the phone over the network (obviously).

In the Android example, do you see the map and the pin at the location of the drone?

JonasVautherin, 2020-04-06:
Any news about that?

Mark718real, 2020-04-27:
Could you write a guide on how to connect it to the SITL jmavsim? I wrote in BACKEND_IP_ADDRESS ip of network and added jmavsim port (4560). App is opened, but "takeoff" button does not work.

JonasVautherin, 2020-04-27:
@Mark718real  seems like your issue is not "failed to upload the mission". Would you mind opening a new issue for your question? :blush:

closed 2022-09-26 by divyanshu1234