mavlink/MAVSDK-Java #33: Connecting android example to SITL jmavsim

Mark718real, 2020-04-27:
Hello, I am a beginner. I try to run the android example and connect it to the SITL jmavsim. I wrote in "BACKEND_IP_ADDRESS" IP of network and added jmavsim port (4560). App is opened, but "takeoff" button does not work. How to connect to jmavsim correctly?

JonasVautherin, 2020-04-27:
There are two ways of doing that:

1. Run `mavsdk_server` on the Android phone, and listen to MAVLink messages coming from the (simulated) drone
2. Run `mavsdk_server` on the computer, next to SITL, and have the Android phone connect to `mavsdk_server`

`BACKEND_IP_ADDRESS` is for the second case. But I believe you want the first one. In the first case, you need to tell jmavsim to send the MAVLink messages to your phone (the usual UDP broadcast will probably not work).

So if I'm right (you're trying 2 above), then you should not change the android-client, but you should point jmavsim to your phone.

An easy way to try that (if you use docker) is to use [`px4-gazebo-headless`](http://github.com/jonasvautherin/px4-gazebo-headless). It would just require you to run:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.10.1 <ip_of_your_phone>
```

with `<ip_of_your_phone>` being something like `192.168.1.12`

Mark718real, 2020-04-30:
Thanks, in px4-gazebo-headless, it works! But I think I should try Jmavsim too.

JonasVautherin, 2020-05-01:
That works too, but you need to make jMAVSim send the MAVLink packets to your phone (which px4-gazebo-headless does for you).

closed 2020-05-01 by JonasVautherin

Mark718real, 2020-05-20:
Hello JonasVautherin
I am a little bit confused about connecting to the jMAVSim. Where I should write the port to connect it to the phone?

julianoes, 2020-05-20:
@Mark718real check these:
https://dev.px4.io/master/en/simulation/#enable-mavbroadcast
https://dev.px4.io/master/en/simulation/#modify-configuration-for-external-broadcasting

Mark718real, 2020-05-21:
As understood I should set MAV_BROADCAST to 1 and write "mavlink start -t <IP of the phone>? I apologize for the misunderstanding

julianoes, 2020-05-25:
@Mark718real you can do either one. You can use `MAV_BROADCAST` to 1 to broadcast to the network and whoever connects to it first will grab it, or you can explicitly set the IP (e.g. in an office with multiple QGCs open potentially connecting!).