mavlink/MAVSDK-Java #35: Signing Mavlink messages with MavSDK

rogersanick, 2020-05-13:
Hi there,

Hope you are well. I'm relatively new to the MavSDK-Java and am really enjoying it so far.

This may be a poorly phrased question - but it is my understanding that with the MavLink 2.0 wire protocol supports message signing - https://mavlink.io/en/guide/message_signing.html

Am I able to facilitate this using the SDK? Please let me know.

Cheers,

Nick

JonasVautherin, 2020-05-13:
Hello! :blush: 

Unfortunately, MAVLink encryption is not (yet) supported in MAVSDK. We have been looking into TLS/dTLS encryption, and the next step there is to move from LibreSSL to OpenSSL in [MAVSDK](https://github.com/mavlink/mavsdk).

If you can and would like to help in that effort, that would be gladly appreciated!

julianoes, 2020-05-14:
@JonasVautherin there is also the [MAVLink signing](https://mavlink.io/en/guide/message_signing.html) capability though as @rogersanick writes. The issue there is that it's not implemented or used in PX4 (see [PR](https://github.com/PX4/Firmware/pull/7530)), and not added to QGroundControl (as far as I can tell from this [PR](https://github.com/mavlink/qgroundcontrol/pull/5582)).
So it's a bit of a chicken and egg problem. If none of the other players have it I'm not very keen to add support for it in MAVSDK :thinking:.

rogersanick, 2020-05-14:
@julianoes - thanks very much for your comment.

To add some color here, I'm hoping to integrate with [Corda](https://www.corda.net/) (a blockchain platform produced by R3). I've been able to do so very generally with MAVsdk (which is awesome), but in order to move forward mavlink signing would be extremely helpful.

julianoes, 2020-05-14:
@rogersanick ok, one way to move it forward would be to get the PX4 PR working and merged.

JonasVautherin, 2020-05-14:
Oh, my mistake, sorry :confused:

julianoes, 2021-11-29:
Closing this. If/when signing is available in PX4 we can reconsider this. It would then happen over int he [MAVASDK C++ core](https://github.com/mavlink/MAVSDK) and not in the Java wrapper.

davidbuzz, 2023-02-07:
https://github.com/davidbuzz/MAVAgent has full signing capabilities, in javascript, its ridiculous that px4 and qgroundcontrol don't have it.

JonasVautherin, 2023-02-07:
Thanks for the constructive comment @davidbuzz! I'm absolutely certain that PX4 and QGroundControl will be happy to take your contributions, though not in Javascript :wink:

davidbuzz, 2023-02-08:
I was just pointing out that there are existing implementations in at least 3 languages... javascript,python,C# , probably more

> https://github.com/davidbuzz/MAVAgent has full signing capabilities, in javascript, its ridiculous that px4 and qgroundcontrol don't have it.

julianoes, 2023-02-09:
Also taking contributions to enable signing for MAVSDK given we can implement and test it against ArduPilot instead. :smile: