mavlink/MAVSDK-Java #36: Adding the virtual joysticks to the android app

Mark718real, 2020-05-20:
Hello everyone, I am new at this. But I would like to add an opportunity to control the drone through the joysticks. Can I do it through the MAVSDK Java? And could someone tell the guideline of how should I implement it?  What data should I work with?

julianoes, 2020-05-20:
You can already do this for offboard mode.

However, if you want to use virtual sticks with a joystick (like it is possible with QGC) we need to add a plugin which sends [MANUAL_CONTROL](https://mavlink.io/en/messages/common.html#MANUAL_CONTROL).

These are the steps to implement this:
1. Add the required API to https://github.com/mavlink/MAVSDK-Proto
2. Implement the C++ MAVLink part.
3. Use the auto-generation to generate this functionality for other languages such as Java.

Some of these things are quite new, so the the docs are a bit sparse. If you're interested we can try to guide you and try to write the docs as we go.

JonasVautherin, 2020-05-20:
What Julian says ^.

If you are interested in adding the MANUAL_CONTROL message, I would see that in a plugin dedicated to the remote controller (maybe `rc.proto`?).

Kutbip, 2020-12-22:
Hello everyone,
I am exploring mavsdk with android. I want to control the drone with Joysticks.
The flow of the application should be like this:
Joysticks and RC transmitter connected with an Android device using USB.
That sends commands to FMU to control the drone.
Here, I have 2 questions:
1. How do I connect or detect multiple drones with USB transmission.
2. Do mavsdk supports different commands (like left, right, up, down) to control drones using Joysticks.

JonasVautherin, 2020-12-24:
MAVSDK now has a `manual_control` plugin, that would do what you want. I think it is available in MAVSDK-Java (otherwise we would just need to update the generated code). Did you try it already?

Kutbip, 2020-12-28:
I have tried to explore in MAVSDK-Java but didn't find the manual control.
I can use system.getAction().land().subscribe(); (takeoff, return to launch).
@JonasVautherin

JonasVautherin, 2021-01-04:
So you're saying that you don't have `system.getManualControl()`? Are you on the latest version? :thinking:

Kutbip, 2021-01-05:
> So you're saying that you don't have `system.getManualControl()`? Are you on the latest version? 🤔

I am using 'io.mavsdk:mavsdk:0.5.1' and 'io.mavsdk:mavsdk-server:0.5.0'
@JonasVautherin 

How to connect multiple drones?
to connect multiple drones is like to connect with multiple mavlink servers?
I can create multiple drones from the QGroundControl app which I want to detect & connect the multiple drones using mavsdk.

JonasVautherin, 2021-01-05:
Oh right, it is there, but not in the `System` helper yet. I need to change that. In the meantime, you can instantiate it separately. For instance:

```java
    System drone = new System();
    ManualControl manualControl = new ManualControl();
```

Then you can use `manualControl` just like if it was `drone.getManualControl()`. Two notes:

* This is temporary, when I release 0.6.0, you'll be able to use `drone.getManualControl()` like for all other plugins.
* If you create your `System` with other arguments (e.g. `new System(host, port)`), you should use the same ones when you create the `ManualControl` plugin (i.e. `new ManualControl(host, port)`).

I hope this helps.

Kutbip, 2021-01-07:
> @Kutbip: I released 0.6.0, it will take a few hours before it is available on MavenCentral. I would be glad if you could try it and let me know if it works 🙏.
> 
> With this new version, `drone.getManualControl()` should just work.

Great.. Now I can access system.getManualControl()
Thanks @JonasVautherin

JonasVautherin, 2021-01-07:
Awesome!

closed 2021-01-07 by JonasVautherin