mavlink/MAVSDK-Java #42: Android Studio Build/Run Errors

simon-xisystems, 2020-08-05:
Hi, 

I am trying to build a demo Android application using the MAVSDK-Java and appear to be running into an issue when trying to build and run the demo android-client. 

The android-client demo app builds fine, but throws the following exception;
`E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.mavsdk.androidclient, PID: 32546
    java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/io.mavsdk.androidclient-jUHyjA6idTc61v6a7rECHw==/base.apk"],nativeLibraryDirectories=[/data/app/io.mavsdk.androidclient-jUHyjA6idTc61v6a7rECHw==/lib/x86, /data/app/io.mavsdk.androidclient-jUHyjA6idTc61v6a7rECHw==/base.apk!/lib/x86, /system/lib, /vendor/lib]]] couldn't find "libnative_lib.so"
        at java.lang.Runtime.loadLibrary0(Runtime.java:1011)
        at java.lang.System.loadLibrary(System.java:1657)
        at io.mavsdk.mavsdkserver.MavsdkServer.<clinit>(MavsdkServer.java:6)
        at io.mavsdk.androidclient.MapsActivity.<init>(MapsActivity.java:56)
        at java.lang.Class.newInstance(Native Method)
        at android.app.Instrumentation.newActivity(Instrumentation.java:1173)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2708)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2892)
        at android.app.ActivityThread.-wrap11(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1593)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6541)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)`

I get a common error when trying to build a new application following the guide, can you confirm how to get around/fix this? 

Thanks

Simon.

simon-xisystems, 2020-08-06:
Just checked and bug isn't present in build 0.3.1

JonasVautherin, 2020-08-11:
Are you trying to run it in the Android emulator with an x86 or x86_64 arch? Unfortunately, we only support armv7 and arm64 on Android.

julianoes, 2020-08-24:
@simon-xisystems bump, could you answer the question above?

simon-xisystems, 2020-08-24:
Sorry gents, yes was trying to run in the emulator, might be worth adding a note to the repo!

JonasVautherin, 2020-08-24:
> might be worth adding a note to the repo!

Sure! Would you mind opening a PR to edit the README? :blush:

simon-xisystems, 2020-08-30:
note added to Readme

closed 2020-08-30 by simon-xisystems