mavlink/MAVSDK-Java #43: How do I connect to "JAVA"? label:question

KyleDemian, 2020-08-11:
Hello, i hava a question

JAVA's running a connection test.
The error below came out.

I don't know what the error is.

FRAME_SIZE_ERROR: 854643
INTERNAL: error in frame handler

I'm using this now. "mavsdk:0.4.0."

`"io.grpc.StatusRuntimeException: INTERNAL: error in frame handler
	at io.grpc.Status.asRuntimeException(Status.java:532)
	at io.grpc.stub.ClientCalls$StreamObserverToCallListenerAdapter.onClose(ClientCalls.java:434)
	at io.grpc.PartialForwardingClientCallListener.onClose(PartialForwardingClientCallListener.java:39)
	at io.grpc.ForwardingClientCallListener.onClose(ForwardingClientCallListener.java:23)
	at io.grpc.ForwardingClientCallListener$SimpleForwardingClientCallListener.onClose(ForwardingClientCallListener.java:40)
	at io.grpc.internal.CensusStatsModule$StatsClientInterceptor$1$1.onClose(CensusStatsModule.java:699)
	at io.grpc.PartialForwardingClientCallListener.onClose(PartialForwardingClientCallListener.java:39)
	at io.grpc.ForwardingClientCallListener.onClose(ForwardingClientCallListener.java:23)
	at io.grpc.ForwardingClientCallListener$SimpleForwardingClientCallListener.onClose(ForwardingClientCallListener.java:40)
	at io.grpc.internal.CensusTracingModule$TracingClientInterceptor$1$1.onClose(CensusTracingModule.java:397)
	at io.grpc.internal.ClientCallImpl.closeObserver(ClientCallImpl.java:459)
	at io.grpc.internal.ClientCallImpl.access$300(ClientCallImpl.java:63)
	at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl.close(ClientCallImpl.java:546)
	at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl.access$600(ClientCallImpl.java:467)
	at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1StreamClosed.runInContext(ClientCallImpl.java:584)
	at io.grpc.internal.ContextRunnable.run(ContextRunnable.java:37)
	at io.grpc.internal.SerializingExecutor.run(SerializingExecutor.java:123)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.io.IOException: FRAME_SIZE_ERROR: 854643
	at io.grpc.okhttp.internal.framed.Http2.ioException(Http2.java:590)
	at io.grpc.okhttp.internal.framed.Http2.access$200(Http2.java:47)
	at io.grpc.okhttp.internal.framed.Http2$Reader.nextFrame(Http2.java:137)
	at io.grpc.okhttp.OkHttpClientTransport$ClientFrameHandler.run(OkHttpClientTransport.java:1024)
	... 3 common frames omitted
`

JonasVautherin, 2020-08-11:
What code are you running? Were you able to run a simple test (e.g. takeoff and land), or is it the first time you use mavsdk-java?

KyleDemian, 2020-08-11:
Hello, :)
yes, I'm using it for the first time.
So I'm having a hard time using 

I'm trying to monitor drone connections and camera gimbal activity and calibration

JonasVautherin, 2020-08-11:
Are you trying to run the [Java examples](https://github.com/mavlink/MAVSDK-Java/tree/master/examples/java-client)? Did you follow the instructions there?

It is difficult for me to understand what's happening without having a clue about what code you are running :slightly_smiling_face: .

KyleDemian, 2020-08-12:
If it doesn't work after more try to do, I'll ask you again!

Thank you for your reply :)

KyleDemian, 2020-08-12:
I have one more question!

I am using pxhawk2,
By the way, the documents say everything is done in pxhawk4,
so is it not available in pxhawk2 ???

I'm more curious because drone control is doing well using mission planner.

JonasVautherin, 2020-08-12:
I think that you are confusing "PX4" and "Pixhawk". Pixhawk is hardware (it's a board), on which you can install software (the "firmware"). PX4 is a firmware, just like ArduPilot. You can install PX4 on a Pixhawk (whichever it is), and you can install ArduPilot on the same Pixhawk, too.

MAVSDK is developped and tested against PX4, meaning that some work is required for it to work with ArduPilot. However, MAVSDK doesn't care about the hardware: PX4 can run on a Pixhawk, on a raspberry pi, or on your computer, as far as MAVSDK is concerned.

Because you use Mission Planner, I guess that you are using ArduPilot. In that case, unfortunately I don't think that you will be able to use MAVSDK.

I hope this helps :blush:

KyleDemian, 2020-08-13:
I was able to straighten out the part I was wrong about.

Thanks you for your help :-)

closed 2020-08-13 by JonasVautherin