mavlink/MAVSDK-Java #47: TakeoffAndLand crash in Andriod

divyanshu1234, 2020-09-04:
I am trying to implement the TakeoffAndLand example on Android. I am able to control the copter using my phone. It is just that when I press the takeoff or land button while the execution of the previous command hasn't completely finished, I get a command denied error and the app crashes. Is there an elegant way built into the SDK to prevent this from happening?

I have added the log messages below:

```
2020-09-04 11:15:54.728 25266-25301/com.example.app E/AndroidRuntime: FATAL EXCEPTION: grpc-default-executor-1
    Process: com.example.app, PID: 25266
    io.reactivex.exceptions.OnErrorNotImplementedException: COMMAND_DENIED: Command Denied
        at io.reactivex.internal.observers.EmptyCompletableObserver.onError(EmptyCompletableObserver.java:51)
        at io.reactivex.internal.operators.completable.CompletableConcatArray$ConcatInnerObserver.onError(CompletableConcatArray.java:60)
        at io.reactivex.internal.operators.completable.CompletableCreate$Emitter.tryOnError(CompletableCreate.java:90)
        at io.reactivex.internal.operators.completable.CompletableCreate$Emitter.onError(CompletableCreate.java:76)
        at io.mavsdk.action.Action$1.onNext(Action.java:216)
        at io.mavsdk.action.Action$1.onNext(Action.java:210)
        at io.grpc.stub.ClientCalls$StreamObserverToCallListenerAdapter.onMessage(ClientCalls.java:421)
        at io.grpc.ForwardingClientCallListener.onMessage(ForwardingClientCallListener.java:33)
        at io.grpc.ForwardingClientCallListener.onMessage(ForwardingClientCallListener.java:33)
        at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1MessagesAvailable.runInContext(ClientCallImpl.java:519)
        at io.grpc.internal.ContextRunnable.run(ContextRunnable.java:37)
        at io.grpc.internal.SerializingExecutor.run(SerializingExecutor.java:123)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: io.mavsdk.action.Action$ActionException: COMMAND_DENIED: Command Denied
        at io.mavsdk.action.Action$1.onNext(Action.java:216) 
        at io.mavsdk.action.Action$1.onNext(Action.java:210) 
        at io.grpc.stub.ClientCalls$StreamObserverToCallListenerAdapter.onMessage(ClientCalls.java:421) 
        at io.grpc.ForwardingClientCallListener.onMessage(ForwardingClientCallListener.java:33) 
        at io.grpc.ForwardingClientCallListener.onMessage(ForwardingClientCallListener.java:33) 
        at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1MessagesAvailable.runInContext(ClientCallImpl.java:519) 
        at io.grpc.internal.ContextRunnable.run(ContextRunnable.java:37) 
        at io.grpc.internal.SerializingExecutor.run(SerializingExecutor.java:123) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:764) 
```

Thank you.

JonasVautherin, 2020-09-04:
Did you try running the java-client example [here](https://github.com/mavlink/MAVSDK-Java/blob/master/examples/java-client/src/main/java/io/mavsdk/example/TakeoffAndLand.java)? It should handle the error.

The idea is that `subscribe()` takes an `onCompleted()` and `onError()` as arguments, as you can see in the example. That's how RxJava works (here you're dealing with a `Completable`).

divyanshu1234, 2020-09-04:
It worked perfectly. Thank you.

closed 2020-09-04 by divyanshu1234