mavlink/MAVSDK-Java #48: goToLocation causing landing

JessicaL237, 2020-09-29:
I've got the drone taking off, and then I'm calling the goToLocation action to get the drone to fly to a specified latitude and longitude. Instead, the drone lands - mavsdk detects a landing. No flying to the location, just landing. 

My code:
`drone.getAction()
			.gotoLocation(waypoint.latitudeInDigitalDegrees, waypoint.longitudeInDigitalDegrees, (float) waypoint.altitudeInMeters, Float.NaN)
			.doOnError(throwable -> {
				LOGGER.error("Failed to go to next waypoint: " + ((Action.ActionException) throwable).getCode());
				success.set(false);
			}).subscribe();`

Any information would be greatly appreciated!

JonasVautherin, 2020-09-29:
What altitude did you set? I'm asking because the altitude is expected above sea level. So if you are flying, say, in Switzerland, and set the altitude to 50m (hoping for 50m above ground), then that will be at least a few hundred meters under your position :innocent:. The altitude given to `gotoLocation` in this case should be `[ground altitude above sea level] + 50`.

hamishwillee, 2020-09-29:
@JonasVautherin I'm sure we had a discussion on making this API more transparent recently :-)

JessicaL237, 2020-09-30:
Yes, that was it thanks! Is there an equivalent that uses relative altitude, or an API call to get ground altitude above sea level so I could factor it in?

JonasVautherin, 2020-09-30:
The problem is that MAVLink in itself does not know the ground altitude. You would need an elevation model for that, and more or less accuracy depending on your use-case.

We can say "goto this altitude w.r.t. the altitude at which the drone took off" because that we know. If this is not accurate enough in your use-case, you need to get an elevation model and use the ground level from there, or something along those lines.

Does that make sense?

JessicaL237, 2020-09-30:
I think so, thanks

closed 2020-09-30 by JessicaL237

mfran89, 2024-08-13:
hypothetically if the drone is doing a mission and the gotolocation function is called will it exit out of auto mode and use the gotolocation wp as its final/landing wp?

julianoes, 2024-08-13:
Please don't necrobump @mfran89. Create a new issue and give enough context.

mfran89, 2024-08-13:
Sorry , I realized that and also decided to go another route !