mavlink/MAVSDK-Java #55: newbie question : controlling aux (pwm 6-16)

zeevmindali, 2020-12-27:
Dear all,

I started to learn on this sdk, mange to figure out most of question.
how can I control with Java, the pwm6-16, I mean, I want to send a signal to servo, for open/close the hook.

tried to look for it, didn't found it :(

zeev

JonasVautherin, 2021-01-04:
Does that exist in [MAVSDK](https://github.com/mavlink/MAVSDK) already? What MAVLink messages would you expect to use?

zeev-mindali, 2022-01-30:
RC9_FUNCTION
And wryie values 1100-1900

julianoes, 2022-01-31:
PX4 or ArduPilot is the first thing that I need to know.

zeev-mindali, 2022-01-31:
ArduPilot , using mission planner

JonasVautherin, 2022-01-31:
So MAVSDK implements the `set_actuator` function.

It is using the MAVLink [`MAV_CMD_DO_SET_ACTUATOR`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR) command here: https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L462-L492

My feeling is that if you set the index to `9` and the value to whatever you need, it should be sent correctly, as implemented [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L487).

zeev-mindali, 2022-01-31:
didn't find this option in java

zeev-mindali, 2022-01-31:
It adopted only in px4 not on ardu

JonasVautherin, 2022-01-31:
You don't have `setActuator` in the `action` plugin in Java?

Or are you saying that `MAV_CMD_DO_SET_ACTUATOR` is not supported by Ardupilot?

zeev-mindali, 2022-02-01:
Dear friend, 
i found out that i can use command 183 (DO_SET_SERVO) to control, tried via mavproxy all is working.
now i trying to connect via java (both java and android) and getting connection refused.
i using herelink remote device and set the ip and port that it's using.

what am i missing?

zeev-mindali, 2022-02-01:
i see that mavsdk is not for ardupilot :( , will seek for alternative option

julianoes, 2022-02-01:
Yes, ArduPilot support is limited and not so easy to achieve due to the differences in how the MAVLink spec is implemented by the two flight stacks. We are trying to get more aligned in the future but it's small steps.

closed 2022-02-01 by julianoes

JonasVautherin, 2022-02-01:
Is `DO_SET_SERVO` the Ardupilot equivalent to `MAV_CMD_DO_SET_ACTUATOR` then?

zeev-mindali, 2022-02-01:
when i run with mavproxy, i controlled the aux1-8 with desired pwm value

julianoes, 2022-02-02:
> Is DO_SET_SERVO the Ardupilot equivalent to MAV_CMD_DO_SET_ACTUATOR then?

That would be good to know.