mavlink/MAVSDK-Java #61: Library source does not match bytecode

sirsquish, 2021-02-03:
Hi,

I got the gradle/maven build by putting in 
    implementation 'io.mavsdk:mavsdk:0.6.0'
    implementation 'io.mavsdk:mavsdk-server:0.6.0'
into build.gradle 

When I get the action.java file it complains about it being different to the compiled version. Was wondering if the source and build were somehow different? It still seems to work for telemetry but I was having an issue with actions and thought maybe this was the reason ( I could arm the the drone in the sim, but the .andthen() never execute) ie I get "arming", and the sim says it's armed, but then nothing else happens.
`
try {
            System drone = MavlinkService.getInstance().drone.get();
            CountDownLatch latch = new CountDownLatch(1);

            drone.getAction().arm()
                    .doOnComplete(() -> Log.i(TAG, "Arming..."))

                    .doOnError(throwable -> Log.e(TAG,"Failed to arm: "
                            + ((Action.ActionException) throwable).getCode()))

                    .andThen(drone.getAction().takeoff()
                            .doOnComplete(() -> Log.i(TAG, "Taking off..."))
                            .doOnError(throwable -> Log.e(TAG, "Failed to take off: "
                                    + ((Mission.MissionException) throwable).getCode())))
                    .delay(5, TimeUnit.SECONDS)

                    .andThen(drone.getAction().land()
                            .doOnComplete(() -> Log.i(TAG, "Landing..."))
                            .doOnError(throwable -> Log.e(TAG, "Failed to land: "
                                    + ((Mission.MissionException) throwable).getCode())))

                    .subscribe(latch::countDown, throwable -> latch.countDown());

            try {
                latch.await();
            } catch (InterruptedException ignored) {
                // This is expected
            }

            ShowMessage.showMessageShort("Post Request");
        } catch (Exception e) {
            ShowMessage.showErrorMessage(TAG, "CRASH?", e);
        }
    }
`

[image]

JonasVautherin, 2021-02-03:
Where does the library source come from?

MichaelSwan3, 2021-02-04:
It comes from 
implementation 'io.mavsdk:mavsdk:0.6.0'
implementation 'io.mavsdk:mavsdk-server:0.6.0'
Looks like the source is pulled out of jar file and compared with class file.

Maybe the class file is incorrectly decompiled - there might not really be an issue at all.
One way to fix it might be to pull from the git repo and build from there - I'll try this after I've had a good look at the serial issue.

closed 2022-09-23 by JonasVautherin