mavlink/MAVSDK-Java #64: Adding additional metadata to MissionPlan

ajaykarthikr, 2021-02-27:
Is there a way to add custom meta data to the `MissionPlan` that is uploaded to the UAV? In our case, there are two GCS attached to the drone and we want to retrieve not only the `MissionPlan` but also our custom metadata uploaded by the first GCS to the second GCS. Both of them are made using `MAVSDK-Java`. Setting some kind of `JSON` in the form of `String` would also work.

julianoes, 2021-03-01:
@ajaykarthikr I'm currently considering adding a mission API which matches the json files that QGroundControl uses.

ajaykarthikr, 2021-03-01:
Thanks for responding. I have few queries regarding this new proposal.
1. Will we be able to add custom parameters to that Json file that will be uploaded to the drone along with mission data?
2. Will those parameters persist if we download it back in the second GCS?
3. Also, will this be integrated in the coming weeks?

julianoes, 2021-03-03:
1. Custom params as in param1 to param7 for a mission item? If you mean this, yes.
2. They should, then it depends what GCS does with them.
3. Working on it.

closed 2025-08-21 by JonasVautherin