mavlink/MAVSDK-Java #65: Fail to use v0.38.0 when building from source

Katawann, 2021-03-30:
Hello,

The issue I have is when I build the mavsdk_server from source with version v0.38.0-1-2 of mavsdk (I build the mavsdk_server from source because I need to have static library instead of shared). No issue with mavsdk v0.37.0 or version before. The output error I get in Android Studio is:

```
2021-03-30 19:00:48.493 6573-6685/com.modulesaero41 E/AndroidRuntime: FATAL EXCEPTION: create_react_context
    Process: com.modulesaero41, PID: 6573
    java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE" referenced by "/data/app/com.modulesaero41-1/lib/arm64/libmavsdk_server.so"...
        at java.lang.Runtime.loadLibrary0(Runtime.java:989)
        at java.lang.System.loadLibrary(System.java:1562)
        at io.mavsdk.mavsdkserver.MavsdkServer.<clinit>(MavsdkServer.java:6)
        at com.modulesaero41.mavsdk.MavsdkModule.<init>(MavsdkModule.java:43)
        at com.modulesaero41.mavsdk.MavsdkPackage.createNativeModules(MavsdkPackage.java:24)
        at com.facebook.react.ReactPackageHelper.getNativeModuleIterator(ReactPackageHelper.java:42)
        at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.java:42)
        at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1347)
        at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1318)
        at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:1225)
        at com.facebook.react.ReactInstanceManager.access$1100(ReactInstanceManager.java:131)
        at com.facebook.react.ReactInstanceManager$5.run(ReactInstanceManager.java:1016)
        at java.lang.Thread.run(Thread.java:761)
```

I guess the issue can also come from the integration I do into a react-nativ app. I'll try to figure it out on my side and I will come with more accurate description if nobody faced this situation so far.

JonasVautherin, 2021-03-30:
Can you show how you build it? The cmake commands?

JonasVautherin, 2021-03-31:
I don't think that you want to change this line. But I don't really understand what the issue is. Doesn't the MavenCentral dependency work in your app? I don't know react-native :sweat_smile:

Katawann, 2021-03-31:
MavenCentral dependency provides a shared library of `mavsdk_server`. There is an [issue](https://github.com/facebook/react-native/issues/29377) with React-Native when we use `libc++_shared.so` and this is issue also happen with MAVSDK. The only workaround I found is to get static library for `mavsdk_server`.

I guess the best is to build a pure Android app and see if the issue still happen

JonasVautherin, 2021-03-31:
> MavenCentral dependency provides a shared library of mavsdk_server.

But you are, too, by changing this line.

There are two things:

* The JNI glue to expose `mavsdk_server` from Java. It is compiled by cmake through gradle
* `libmavsdk_server.so`, that comes from the [releases](https://github.com/mavlink/mavsdk/releases).

If you want to use a static libc, you have to recompile `libmavsdk_server.so` as well :+1:.

Katawann, 2021-04-01:
ok so I guess I'm lucky it is working. I'm closing the issue and will try your proposition and see if I still have the same problem. Thank you !

divyanshu1234, 2021-04-02:
I just updated the dependencies to `0.8.0` and I too am getting the same error. I'm not using react-native.

```
def mavsdk_version = "0.8.0"
implementation "io.mavsdk:mavsdk:$mavsdk_version"
implementation "io.mavsdk:mavsdk-server:$mavsdk_version"
```

```
Caused by: java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE" referenced by "/data/app/<package>-gPP2YdUfzajwPSPZR8usKw==/base.apk!/lib/arm64-v8a/libmavsdk_server.so"...
        at java.lang.Runtime.loadLibrary0(Runtime.java:1016)
        at java.lang.System.loadLibrary(System.java:1669)
        at io.mavsdk.mavsdkserver.MavsdkServer.<clinit>(MavsdkServer.java:6)
```

Things work perfectly fine with `0.7.0`

JonasVautherin, 2021-04-02:
Aha! I think I did not update the NDK correctly, and this is a feature coming with C++17. Let me have a look!

JonasVautherin, 2021-04-02:
I am not completely sure, but I believe that this happens because you are using an Android NDK older than r22. Could it be?

The `mavsdk_server` package links libc as a shared library (which @Katawann wants to change), but then I would think that libc is coming from the final app, and hence from your NDK. For @divyanshu1234, I believe that updating the NDK may just do it. Could you please try?

For @Katawann, if you want to link libc statically, then you'll have to do that when building `mavsdk_server`, and if you use dockcross (like the MAVSDK CI), then I think it will need to wait until [dockcross is updated](https://github.com/dockcross/dockcross/pull/479).

Please let me know how it goes!

JonasVautherin, 2021-04-02:
For the record, I believe that [this](https://github.com/android/ndk/issues/609) is related

divyanshu1234, 2021-04-03:
> I believe that this happens because you are using an Android NDK older than r22. Could it be?

I was using NDK `22.0.7...` earlier. Even after updating it to `22.1.7...` I'm getting the same problem.

Is it possible that you are using platform headers that are higher than `android-28` to build the library? That's probably why it isn't running on my test device.
I'll try to run it on newer versions of android to see if it still gives a problem there.

[This](https://stackoverflow.com/questions/27091001/how-to-use-mkfifo-using-androids-ndk/27093163#27093163) answer may be of help.

JonasVautherin, 2021-04-03:
> Is it possible that you are using platform headers that are higher than android-28 to build the library?

You mean [those values](https://github.com/mavlink/MAVSDK-Java/blob/main/mavsdk_server/build.gradle#L122-L123)?

```
minSdkVersion 21
targetSdkVersion 29
```

My intuition is that `"_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE"` is a libc symbol (coming with C++17), and therefore I would expect the problem to come from the NDK version. Also, in [my link above](https://github.com/android/ndk/issues/609) they talk about filesystem support coming with NDK 22. And I could reproduce the problem with an older NDK, and then it worked when I updated to NDK 22.

Now, maybe it also requires `libmavsdk_server.so` to be built with NDK 22. Let me try again...

divyanshu1234, 2021-04-03:
> I'll try to run it on newer versions of android to see if it still gives a problem there.

The same problem persists even with API level `29`.

> You mean [those values](https://github.com/mavlink/MAVSDK-Java/blob/main/mavsdk_server/build.gradle#L122-L123)?

I meant something like [this](https://github.com/dropbox/djinni/blob/4f3aa6971b2b539cfadec4df2243a268651c4de3/example/android/app/jni/Application.mk#L6).

> Now, maybe it also requires `libmavsdk_server.so`

Yes, I was thinking the same. It had skipped my mind that `libmavsdk_server.so` was added externally and not built using NDK. (Please correct me if I'm wrong)

JonasVautherin, 2021-04-03:
I think that `mavsdk-server` needs to be built with NDK 22 (and I failed that for `mavsdk-server:0.8.0`). I am publishing a new version `mavsdk-server:0.8.1` which will be out on MavenCentral in a few hours.

In the meantime, if you want to try, you can try building it yourself: Build the [android-client](https://github.com/mavlink/MAVSDK-Java/tree/main/examples/android-client) example after you have uncommented [this line](https://github.com/mavlink/MAVSDK-Java/blob/main/examples/android-client/settings.gradle#L11) (so that it will fetch `mavsdk:0.8.0` from MavenCentral, and it will build `mavsdk-server` using your NDK 22).

divyanshu1234, 2021-04-04:
> I am publishing a new version `mavsdk-server:0.8.1` which will be out on MavenCentral in a few hours.

`mavsdk-server:0.8.1` is working perfectly.

Thank you..

JonasVautherin, 2021-04-04:
Awesome! @Katawann would you mind testing with `mavsdk-server:0.8.1` as well?

Katawann, 2021-04-05:
mmmh it's still failing to run with error 
` java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE" referenced by "/data/app/com.modulesaero41-2/lib/arm64/libmavsdk_server.so"...`. 

I'll try to find some time to do the modifications you proposed to build `libmavsdk_server` as well

JonasVautherin, 2021-04-05:
@Katawann using the shared libc?

Katawann, 2021-04-06:
If I use the shared libc, I failed back to the issue I have because I have multiple libc++_shared.so:

```
More than one file was found with OS independent path 'lib/x86/libc++_shared.so'. If you are using jniLibs and CMake IMPORTED targets, 
see https://developer.android.com/studio/preview/features#automatic_packaging_of_prebuilt_dependencies_used_by_cmake
```

I gave more details few weeks ago in a [post](https://discuss.px4.io/t/failed-to-run-mavsdk-server-in-react-native-app-on-android/20607) on the PX4 discussion forum

JonasVautherin, 2021-04-06:
Ok, I'm sorry I'm not really active on discuss :innocent:.

If you don't mind, I'll close this issue (that became dedicated to the `filesystem` bug) and we can continue in MAVSDK: https://github.com/mavlink/MAVSDK/issues/1397.

closed 2021-04-06 by JonasVautherin

divyanshu1234, 2021-05-14:
> I just updated the dependencies to `0.8.0` and I too am getting the same error. I'm not using react-native.
>
> ```
> def mavsdk_version = "0.8.0"
> implementation "io.mavsdk:mavsdk:$mavsdk_version" [...]

I am facing the same problem with `mavsdk-server:0.9.0`

> I think that `mavsdk-server` needs to be built with NDK 22 (and I failed that for `mavsdk-server:0.8.0`).

@JonasVautherin do you think it's the same problem as before? If yes, can you please publish the fixed version?

JonasVautherin, 2021-05-14:
Hmm could be, though I thought that I had updated dockcross to use NDK22 :thinking:. I'll have a look ASAP!

JonasVautherin, 2021-05-14:
I pushed 0.9.1, let me know if that's better :-). If yes, I think I found a way to solve that once and for all :+1:

divyanshu1234, 2021-05-15:
Just tried and tested it. Works great.

Thank you.

JonasVautherin, 2021-05-15:
Great, thanks for testing! FYI, the required ndkVersion is now set [here](https://github.com/mavlink/MAVSDK-Java/blob/main/mavsdk_server/build.gradle#L154), so I should not forget it anymore :wink:

nachod30, 2021-06-28:
Hello, I have the same error with react native app (dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE"), I have compiled aar mavsdk server 0.9.0 with c++_static flag. Where can I found 0.9.1 version of mavsdk server? Thanks.

nachod30, 2021-06-28:
> I pushed 0.9.1, let me know if that's better :-). If yes, I think I found a way to solve that once and for all

Hello, I have the same error with react native app (dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE"), I have compiled aar mavsdk server 0.9.0 with c++_static flag. Where can I found 0.9.1 version of mavsdk server? Thanks.

julianoes, 2021-06-28:
```
echo "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE" | c++filt
std::__ndk1::__fs::filesystem::__status(std::__ndk1::__fs::filesystem::path const&, std::__ndk1::error_code*)
```

Where did you get the mavsdk_server binary from?

JonasVautherin, 2021-06-28:
> I have compiled aar mavsdk server 0.9.0 with c++_static flag. Where can I found 0.9.1 version of mavsdk server?

That's the issue, you need 0.9.1. It is available with maven: https://mvnrepository.com/artifact/io.mavsdk/mavsdk-server.

It you want to build it, just take the latest state [here](https://github.com/mavlink/MAVSDK-Java/tree/main/mavsdk_server), I would say.

nachod30, 2021-06-28:
@julianoes I compiled the main branch binary and it gave the above error. (https://github.com/mavlink/MAVSDK-Java/tree/main/mavsdk_server)

@JonasVautherin I just tried 0.9.1 , adding in gradle:
`implementation group: 'io.mavsdk', name: 'mavsdk-server', version: '0.9.1'`

and it gave me an old error:
`More than one file was found with OS independent path 'lib/x86/libc++_shared.so'. If you are using jniLibs and CMake IMPORTED targets...`

Any ideas or suggestions?
Thanks.

JonasVautherin, 2021-06-28:
I think that this is similar to what @Katawann was doing. Would you mind opening a new issue for that and tag @Katawann there?

divyanshu1234, 2022-06-22:
This problem has returned when I tried to add the VLC Android dependency:

```
java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE" referenced by "/data/app/~~_P1dshRgTSb_pMXk3WYdzw==/com.urbanmatrix.android.launchpad-MykqlwyrHRV-o-L0UFgDvQ==/base.apk!/lib/arm64-v8a/libmavsdk_server.so"...
```

The problem came earlier because MAVSDK-Java wasn't using the latest NDK. VLC apparently is using NDK 24 ([here](https://code.videolan.org/videolan/vlc-android/-/issues/2518)).

julianoes, 2022-08-04:
@divyanshu1234 so we should update the NDK version for mavsdk?

divyanshu1234, 2022-08-05:
I did some tests and used a release with the updated NDK, and the problem persisted. I noticed that this problem comes only with the VLC NDK. It may be that it's a problem from their side only.

But we should shift to the LTD NDK anyway.