mavlink/MAVSDK-Java #68: `LogFiles#getDownloadLogFiles` method doesn't work as expected

divyanshu1234, 2021-06-11:
This is the signature of the download log file method in log_files package in the Java version:
```
public Flowable<LogFiles.ProgressData> getDownloadLogFile()
```
And this is the corresponding one in Python:
```
async def download_log_file(self, entry, path):
```

The Java one doesn't have any parameters for entry and path and always throws `FILE_OPEN_FAILED: File Open Failed` error when I try to use it.

julianoes, 2021-06-11:
Thanks for reporting @divyanshu1234. This looks indeed wrong, right @JonasVautherin?

JonasVautherin, 2021-06-14:
Should be fixed in latest `main`. @divyanshu1234 could you give it a try? Then I could make a new release with those changes :+1:

divyanshu1234, 2021-06-14:
I tried downloading a few logs using SITL and it is working fine. I will not be able to try it on hardware this week. But I've hardly ever faced any issue with MAVSDK that emerged using hardware that wasn't there in SITL.

Thanks for the fix.

JonasVautherin, 2021-06-15:
Ok let's close it then, it will come in the next release (I'll try to create a release today or tomorrow)

closed 2021-06-15 by JonasVautherin