mavlink/MAVSDK-Java #70: React Native Module mavsdk-server

nachod30, 2021-06-28:
@Katawann
Hi, I trying to create a React Native Module with Android Java MAVSDK.

I compiled the main branch binary mavsdk-server with c++_static flag and it gave this error: 
"_ZNSt6__ndk14__fs10filesystem8__statusERKNS1_4pathEPNS_10error_codeE"...

I have tried 0.9.1, adding in gradle:
`implementation group: 'io.mavsdk', name: 'mavsdk-server', version: '0.9.1'`

and it gave me an old error:
`More than one file was found with OS independent path 'lib/x86/libc++_shared.so'. If you are using jniLibs and CMake IMPORTED targets...`

Any ideas or suggestions?
Thanks.

_Originally posted by @nachod30 in https://github.com/mavlink/MAVSDK-Java/issues/65#issuecomment-869668685_

JonasVautherin, 2021-06-28:
@Katawann I asked @nachod30 to tag you because I believe you were also running mavsdk-java with react native, so maybe you have some insights?

Katawann, 2021-06-28:
Yes sure, but actually I had no time to fix this issue on my side...

@nachod30 if you built the library like I did (change the flag to static), maybe try previous version of MAVSDK to start with a working version. Change the build.gradle to get previous version. I'm stuck with the version v0.35.1 of MAVSDK but at least it's working :)
`l.22 def mavsdk_server_release = "v0.35.1"
`

`l.128 version '0.6.0'
`

From what I tested, it works to build MAVSDK server statically like this until v0.37.0. From version v0.38.0, it fails as mentioned in the issue #65. Another issue on this subject is opened here #66. @JonasVautherin proposed another way to build the library statically, you might want to try this solution.

Finally, this libc++_shared.so seems to be a known issue in React Native but I just checked and there is still no fix for that https://github.com/facebook/react-native/issues/29377

Let me know if you succeed to build with previous version !

nachod30, 2021-06-29:
Hi, Thank you very much @Katawann. Yes, I succeed building with previous version changing the build.gradle to get previous version! (v0.35.1) This error is gone! 

But now there is a new issue when I do: `int mavsdkServerPort = mavsdkServer.run();` , my app crash without showing errors.

Any suggestions?
Thanks.

Katawann, 2021-06-29:
Somehow this error -6 looks really familiar to me but I can't remember how I fixed it... 

Otherwise we need to dig to generate this mavsdk library statically in a proper way

julianoes, 2022-05-26:
Can this be closed? Or is there still work going on?

closed 2022-09-23 by JonasVautherin