mavlink/MAVSDK-Java #71: Requesting a new release

divyanshu1234, 2021-08-12:
It has been a while since the Java bindings got a new release. The newer versions of MAVSDK have certain features that are quite helpful. @JonasVautherin can you please create a new release with the latest stable MAVSDK code?

JonasVautherin, 2021-08-14:
There is something wrong when building MAVSDK-Java with the latest proto submodule. I'm out next week, I'll have to check that when I'm back. Sorry for the delay :confused:

divyanshu1234, 2021-08-16:
No issues. Please take your time.

divyanshu1234, 2021-08-31:
Any update on this?

JonasVautherin, 2021-09-05:
On its way, I fixed the protos tonight: https://github.com/mavlink/MAVSDK-Proto/pull/254

JonasVautherin, 2021-09-05:
Hmm I am still having an issue when building locally. I'll try to push 0.10.0 to Maven to see if it works. Feel free to test it when it's out (in a few hours) and let me know :+1:

JonasVautherin, 2021-09-05:
It fails in `android-client` with the same error as I got locally :cry::

```
2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:mergeDebugJavaResource'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > More than one file was found with OS independent path 'google/protobuf/field_mask.proto'.
```

I don't understand why, because I did not change the versions in that update... And I did not find a way to fix it yet.

@divyanshu1234: I guess you get the same error in your project? Would you have an idea how to fix this?

divyanshu1234, 2021-09-06:
I had it with some other dependency for which I just added the following to the app: build.gradle

```
android {
//...
  packagingOptions {
        pickFirst '<dependency>'
    }
 }

```

But this apparently did not work for someone on Slack. Let me pull the code and see if I can make it work. I'll get back to you in some time.

JonasVautherin, 2021-09-06:
I tried to play with `packagingOptions`, but I got stuck after:

```
    packagingOptions {
        pickFirst 'google/protobuf/field_mask.proto'
        pickFirst 'google/protobuf/type.proto'
        pickFirst 'google/protobuf/api.proto'
        pickFirst 'google/protobuf/source_context.proto'
        pickFirst 'google/protobuf/empty.proto'
        pickFirst 'google/protobuf/wrappers.proto'
        pickFirst 'google/protobuf/struct.proto'
        pickFirst 'google/protobuf/any.proto'
        pickFirst 'google/protobuf/timestamp.proto'
    }
```

Let me know if you find a way :-)

divyanshu1234, 2021-09-06:
When I tried to build 0.10.0 on my project, I got a number of messages like this:
```
Duplicate class com.google.protobuf.WireFormat$FieldType$1 found in modules jetified-protobuf-javalite-3.12.0 (com.google.protobuf:protobuf-javalite:3.12.0) and jetified-protobuf-lite-3.0.1 (com.google.protobuf:protobuf-lite:3.0.1)

Duplicate class com.google.protobuf.Value$KindCase found in modules jetified-mavsdk-0.10.0 (io.mavsdk:mavsdk:0.10.0) and jetified-protobuf-javalite-3.12.0 (com.google.protobuf:protobuf-javalite:3.12.0)
```

I've had a similar problem while using a library that I had built. There were conflicts because two dependencies were 'leaking' their own dependencies. I replaced all the `api` configurations with `implementation`  and it started working.
Please refer [this](https://stackoverflow.com/questions/44413952/gradle-implementation-vs-api-configuration) StackOverflow answer.

I guess one conflict, in this case, would be the protobuf dependency in mavsdk and grpc that have been used.

Please try this, and in the meantime, I'll try to find a workaround.

divyanshu1234, 2021-09-06:
For the android-client, you can declare the mavsdk dependency as follows:
```
implementation("io.mavsdk:mavsdk:$mavsdk_version") {
    exclude group: 'com.google.protobuf', module: 'protobuf-javalite'
}
```
This seems to be working. But I think replacing all `api` with `implementation` in the library's gradle would prevent the need for this workaround.

Apart from this, we're also using mavsdk-java for a non-android application, and there are a few things not working in that. I don't have my full test setup with me for a few days. But, I'll try to debug it. (The problem could be with my current setup too)

divyanshu1234, 2021-09-06:
I have noticed a weird behaviour. When I start the app first (with the mavsdk server running on it) and start SITL later, the telemetry data streams don't work, but the arm and takeoff commands work. To make it work properly, I need to have SITL already running before I start the app.

I tried the same with the android mavsdk_server:0.9.1 and it was working properly in both cases.

Is this behaviour intentional?

JonasVautherin, 2021-09-06:
> Is this behaviour intentional?

It is not, that's a bug on which I am working :+1:

divyanshu1234, 2021-09-07:
Okay. Thanks.

divyanshu1234, 2021-11-02:
> It is not, that's a bug on which I am working

Hello @JonasVautherin. Were you able to fix this bug? If yes, can you please create a new release for Java?

JonasVautherin, 2021-11-02:
I think it should be fixed in the latest mavsdk_server release. Would you mind trying it?

divyanshu1234, 2021-11-02:
Yes, `mavsdk_server` `v0.46.1` is working perfectly in an external server configuration.

JonasVautherin, 2021-11-02:
Ok I'll cut a new release ASAP

divyanshu1234, 2021-11-02:
Great! Thanks!

divyanshu1234, 2021-11-08:
@JonasVautherin Will you be able to create a release this week?

JonasVautherin, 2021-11-12:
Sorry @divyanshu1234 I've been busy and this required some changes as the underlying API has slightly changed.

Would you mind trying with my changes? If it works, I'll push to maven.

divyanshu1234, 2021-11-13:
> Sorry @divyanshu1234 I've been busy and this required some changes as the underlying API has slightly changed.

@JonasVautherin no issues at all.

> Would you mind trying with my changes?

Sure just give me a day.

PR #78 (2021-11-15): Update to mavsdk_server v0.48.0
@divyanshu1234: Would you mind trying with those changes? If it works I'll push a new artifact 0.11.0 :+1:.

Resolves #71.
closed 2021-11-15 by JonasVautherin