mavlink/MAVSDK-Java #72: Remove replay(1) from streams

divyanshu1234, 2021-09-27:
The replay operator is added to all the flowable streams in the library.

```
return flowable
  .replay(1)
  .autoConnect();
```

This creates certain issues while using them.

1. It is difficult to differentiate between live events and cached ones. For example, if we subscribe to Camera.CaptureInfo at some point in time, the last data point gets sent even if the photo may have been taken 10 minutes ago.

2. If someone is using LiveData on Android, which does its own caching, the caching on the library side further creates some unexpected behaviour.

Yes, the replay is useful under certain situations, but it requires tricky workarounds to invalidate it downstream once its added. Therefore, I feel that simply returning the flowable without the replay is good enough, and the library should let the user implement the caching on his/her own.

Any thoughts?

JonasVautherin, 2021-09-27:
Sounds good to me!

divyanshu1234, 2021-09-27:
Great. I'll create a PR either today or tomorrow.

PR #73 (2021-09-27): Removed last object replay behaviour of streams
Resolves #72 

@JonasVautherin, the code is building, but I don't know why I am not able to use the generated jar file to test the code properly. It's giving me several dependency issues. Though I am able to use the maven dependency (but it has the old code)
closed 2021-09-27 by JonasVautherin