mavlink/MAVSDK-Java #83: Requesting a new release

divyanshu1234, 2022-01-29:
@JonasVautherin can you please create a new Maven release with the latest stable version of MAVSDK?

JonasVautherin, 2022-01-29:
For some reason I had issues building because of some deprecations. I started updating the tooling here: https://github.com/mavlink/MAVSDK-Java/pull/84.

But I have an issue with `./gradlew build`:

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
   > 2 files found with path 'lib/arm64-v8a/libmavsdk_server.so' from inputs:
      - /home/jones/Documents/px4/sdk/mavsdk-java/mavsdk_server/build/intermediates/merged_jni_libs/debug/out/arm64-v8a/libmavsdk_server.so
      - /home/jones/Documents/px4/sdk/mavsdk-java/mavsdk_server/build/intermediates/cxx/Debug/3y4z4f6x/obj/arm64-v8a/libmavsdk_server.so
     If you are using jniLibs and CMake IMPORTED targets, see
     https://developer.android.com/r/tools/jniLibs-vs-imported-targets
```

divyanshu1234, 2022-01-29:
A somewhat hacky solution is to add this to the `build.gradle`.

```
android {
...
    packagingOptions {
        pickFirst '**/libmavsdk_server.so'
    }
...
}
```

I thought that removing the `extractMavsdkServer` task should make it work since the docs stated:

> External native build now automatically packages those libraries, so explicitly packaging the library with jniLibs results in a duplicate. To avoid the build error, move the prebuilt library to a location outside jniLibs or remove the jniLibs configuration from your build.gradle file.

This didn't work. So it looks like for a better solution, we'll need to do something with the external build process. I'm not sure what that is at the moment.

But for the time being, the `pickFirst` method does the trick as I am able to build it using your PR.

JonasVautherin, 2022-01-29:
I extracted them in `src/main/prebuiltLibs` instead of `src/main/jniLibs` and it built, not sure if it runs :sweat_smile:. Would you mind trying it? I'll update the sdk package next :+1:

divyanshu1234, 2022-01-30:
Sure just give me a day. I'll try it with the server V1.0.3

divyanshu1234, 2022-01-30:
It looks like it's working great!

I tried it with `mavsdk-server` `v1.0.8`, using the protos at `5cb33bbe`.

The only problem I found was with `slf4j-api:2.0.0-alpha6`. It is still unstable is creating problems in mavsdk dependants. Please change it to the latest stable `1.7.35`

JonasVautherin, 2022-01-30:
I have issues publishing with the new tooling, I'll check again tomorrow :+1:

JonasVautherin, 2022-01-31:
I published 1.0.0, should be on mavenCentral in a few hours. Let me know if it works :crossed_fingers:

divyanshu1234, 2022-02-01:
Awesome! I'll test it out on hardware today.
Thanks.

divyanshu1234, 2022-02-01:
Looks like something's gone wrong. io/mavsdk/mavsdk/1.0.0 is available on repo.maven and is working properly.

[image]

But it looks like the aar file is missing for mavsdk-server/1.0.0 and hence Gradle isn't able to find it.

[image]

JonasVautherin, 2022-02-01:
Ok I released 1.0.1, would you mind trying it when it's on mavenCentral? Thanks for the help :laughing:

divyanshu1234, 2022-02-01:
I did a few flights with our drones and it looks like it's working great.

> Thanks for the help

Glad to help and thank you for the release.

closed 2022-02-01 by divyanshu1234