mavlink/MAVSDK-Java #85: RC_CHANNELS_OVERRIDE

zeev-mindali, 2022-01-30:
Dear all,

i need to send values to channel 9-16.
i saw that there is RC_CHANNELS_OVERRIDE commands that can do just that.
how i can implement this in java

thanks in advance

JonasVautherin, 2022-01-30:
The API needs to be added to mavsdk-proto first, and then it needs to be implemented in mavsdk-c++. Would you be willing to contribute this in C++? If yes, please move this issue to https://github.com/mavlink/MAVSDK :blush:

zeev-mindali, 2022-01-31:
sorry, as much as i want to help, don't know c++, working with java and android studio
is there another way to achieve it? mavlink message?

julianoes, 2022-01-31:
There needs to be an easier way to just set a servo than using RC_CHANNELS_OVERRIDE. Let's follow up in #55.

closed 2022-01-31 by julianoes