mavlink/MAVSDK-Java #87: mavsdk java Connection refused: connect

zeev-mindali, 2022-02-03:
Dear all,

trying to run the java example with mavsdk.
using latest libraries from git.

when starting the software, getting a connection refused: connect

i using herelink controller, and after digging in , i understand that it's getting udp connection and not a tcp connection (as far as i can tall)

in java command 
<code>
System drone = new System("192.168.50.238",14552);
</code>

i dont see how to set udp connection to it.

once it's solved, i have an sdk (tested on sitl) for controlling apm throw mavsdk

any ideas?

julianoes, 2022-02-03:
Can you lay out what is sending on what port and how you are trying to connect?

JonasVautherin, 2022-02-03:
>  System drone = new System("192.168.50.238",14552); 

This is most likely wrong. If you look at [the docs](https://github.com/mavlink/MAVSDK-Java/blob/main/sdk/mavsdk/src/main/java/io/mavsdk/System.java#L77-L78):

```java

  /**
   * Create a System object.The plugins are initialized lazily, when the corresponding
   * get method is called.
   *
   * @param host the address of mavsdk_server
   * @param port the port of mavsdk_server
   */
  public System(@NonNull String host, int port)
```

You should run [mavsdk_server](https://github.com/mavlink/MAVSDK/releases/tag/v1.0.8) separately:

* `$ ./mavsdk_server -p 50051` will listen for heartbeats from your drone on UDP port 14540, it is the same as running `$ ./mavsdk_server -p 50051 udp://:14540`
* `$ ./mavsdk_server -p 50051 udp://192.168.50.238:14552` will send heartbeats to `192.168.50.238` on port `14552`, so this time it's the drone that is listening on 14552.

Then you should connect your `System` object to `mavsdk_server`, e.g.:

```java
System drone = new System("127.0.0.1", 50051); 
```

If you run `mavsdk_server` on the machine running the java code, with `-p 50051`, then that's the default, so you can just use:

```java
System drone = new System();
```

zeev-mindali, 2022-02-03:
oh, so let me check if i understand correctly
i will create mavsdk_server on my machine and send the message to mavsdk, and then the mavsdk_server will send it to the drone, i get it right?

and if i will use the android platform for herelink, i will create mavsdk on android and give him the connection to udp of the drone?

JonasVautherin, 2022-02-03:
Exactly. For Android, there is a package on maven for mavsdk_server. We use it in the example app here: https://github.com/mavlink/MAVSDK-Java/blob/main/examples/android-client/app/src/main/java/io/mavsdk/androidclient/MapsActivity.java#L92-L93

zeev-mindali, 2022-02-03:
mavsdk_server is connected to my drone, cheers

but when running java examples i getting the error:
<code>
12:58:40: Executing ':SystemShell.main()'...

> Task :compileJava UP-TO-DATE
> Task :processResources NO-SOURCE
> Task :classes UP-TO-DATE

> Task :SystemShell.main()
12:58:41.492 [main] DEBUG io.mavsdk.example.SystemShell - Starting example: system shell
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/C:/Users/zeevm/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protobuf-lite/3.0.1/59b5b9c6e1a3054696d23492f888c1f8b583f5fc/protobuf-lite-3.0.1.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

</code>

my goal is to send a message to the drone, in mavproxy i sent :
<code>
message COMMAND_LONG 0 0 183 0 7 1500 0 0 0 0 0
</code>

i sure i missing something stupied here, but what?

also in android , i have mavserver, but where i configure there the drone ip and port?

i think i almost there

thanks for all the help, very appricated

JonasVautherin, 2022-02-03:
On Android it's here: https://github.com/mavlink/MAVSDK-Java/blob/main/examples/android-client/app/src/main/java/io/mavsdk/androidclient/MapsActivity.java#L93

How do you run the java examples?

zeev-mindali, 2022-02-03:
running the mavserver (found ready exe file :) )
about the java,  i run from intellij ide the system shell.
but it's stuck and rolling for ever

closed 2022-02-04 by zeev-mindali