mavlink/MAVSDK-Java #88: sending commands in java

zeev-mindali, 2022-02-04:
Dear all,

after struggling to connect mavsdk to apm drone via herelink,
i was finally able to connect :)

how do i send commands to the drone?

in mav proxy, i send the command:
<code>
COMMAND_LONG 0 0 183 0 7 1500 0 0 0 0 0
</code>

how can i achieve that with mavsdk?
once i will be able to send this commands, i can finally write a simpler higher level of sdk to implement both px4 and apm for novice users.

JonasVautherin, 2022-02-04:
You don't have access to low-level mavlink messages, just the MAVSDK API. So all the functions defined [here](https://github.com/mavlink/MAVSDK-Proto/tree/main/protos).

For instance, you don't send a command to takeoff, you call `drone.getAction().takeoff().subscribe()`. Did you manage to run one of the examples?

zeev-mindali, 2022-02-04:
dear friend, 

this is not my issue,
i want to control the aux or servo for controlling custom camera.
with mavproxy i was able to send the commands.
is there a way to send the commands with mavsdk?

JonasVautherin, 2022-02-04:
In MAVSDK-C++, you can use the mavlink_passthrough. In MAVSDK-Java, you have to use the API given above.

This said, MAVSDK-Java has the [`set_actuator`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L104) function, which I believe would do what you want in PX4. It uses the [`MAV_CMD_DO_SET_ACTUATOR`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR) mavlink message. Does that not work in Ardupilot?

I think you mentioned [`MAV_CMD_DO_SET_SERVO`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO) in another thread. Is that what is used by Ardupilot? Would you be able to add support for `MAV_CMD_DO_SET_ACTUATOR` into Ardupilot? That way it would work with MAVSDK, maybe.

zeev-mindali, 2022-02-06:
MAV_CMD_DO_SET_SERVO can control PWM values of servo 1-8
in mav proxy it's works, don't know if APM can handle actuator, didn't found it

zeev-mindali, 2022-02-07:
I end up trying now to run mavsdk server for connection and low level (dronefleet.io) for commands. 
if not, I will write entire new sdk for all the commands of apm and px4 wrapper, to upper level.

closed 2022-02-07 by zeev-mindali

Zxr1230, 2023-08-25:
> Dear all,
> 
> after struggling to connect mavsdk to apm drone via herelink, i was finally able to connect :)
> 
> how do i send commands to the drone?
> 
> in mav proxy, i send the command: ` COMMAND_LONG 0 0 183 0 7 1500 0 0 0 0 0 `
> 
> how can i achieve that with mavsdk? once i will be able to send this commands, i can finally write a simpler higher level of sdk to implement both px4 and apm for novice users.

Hello,

Just wonder if you tried using the android-client in MAVSDK-Java/examples/? I am trying to control my ArduPilot drone with this example app installed on the Herelink ground station but it seems unable to run the MAVSDK server. 
Thank you!