mavlink/MAVSDK-Java #97: How android mavsdk-server connect with a gazebo docker container?

zealrussell, 2022-04-25:
Dear all,
I run the `jonasvautherin/px4-gazebo-headless` docker image in my windows laptop, and(1)when I run mavsdk-server and mavsdk-java in the laptop, the mavskd-server automatically connect with gazebo-docker, so it works successfully, but (2) when I run the mavsdk-server and mavsdk-java in my android phone(they use same wifi), the mavsdk-server could not connect with the gazebo-docker. I use the following code to create an android application:
``` 
MavsdkServer mavsdkServer = new MavsdkServer();
int port = mavsdkServer.run(''udp://:14540'');
drone = new System(''127.0.0.1'', port); 
```
After carefully reading `getting-started-with-mavsdk-java/` ,I think the question is `udp://:14540`, so I tried `udp://192.168.1.10:14540` (my laptop ip address) but still didn't work. I even use the commander `docker run --rm -it jonasvautherin/px4-gazebo-headless:1.12.3 192.168.1.10` but still. I really want to know how could I connect them with eachother. My english is bad so I draw the picture [question.png](https://postimg.cc/SnYk29zF). Thanks a lot!

JonasVautherin, 2022-04-25:
Did you find a way?

JonasVautherin, 2022-04-25:
Two things:

1. Why do you use double single-quotes `''` and not double quotes `"` in the code above? Does that even compile? :thinking:
2. With your code above, your Android app is listening on UDP port 14540. You should be able to just point px4-gazebo-headless to it. So instead of using the IP of the laptop, use the IP of the Android device:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.12.3 <ip_of_android_device>
```

zealrussell, 2022-04-25:
For the first, it is a typo, I have used double quotes in my code. Sorry for my mistake. 
For the second, I use `docker run --rm -it jonasvautherin/px4-gazebo-headless:1.12.3 <ip_of_android_device>` and it really works! Thank you for your help. Best wishes!

closed 2022-04-25 by JonasVautherin