mavlink/MAVSDK-Java #99: java.lang.NoClassDefFoundError: com.google.protobuf.GeneratedMessageLite

ankitatcodingnebula, 2022-07-05:
Hello MAVSDK Team,

I updated the Java SDK from 0.8.0 to 1.x and started getting a compilation error on my project

startup failed:

> General error during semantic analysis: java.lang.NoClassDefFoundError: com.google.protobuf.GeneratedMessageLite
> 
> java.lang.RuntimeException: java.lang.NoClassDefFoundError: com.google.protobuf.GeneratedMessageLite
> 	at org.codehaus.groovy.control.CompilationUnit.convertUncaughtExceptionToCompilationError(CompilationUnit.java:1118)
> 	at org.codehaus.groovy.control.CompilationUnit.applyToPrimaryClassNodes(CompilationUnit.java:1098)
> 	at org.codehaus.groovy.control.CompilationUnit.doPhaseOperation(CompilationUnit.java:640)
> 	at org.codehaus.groovy.control.CompilationUnit.processPhaseOperations(CompilationUnit.java:618)
> 	at org.codehaus.groovy.control.CompilationUnit.compile(CompilationUnit.java:595)
> 	at org.codehaus.groovy.control.CompilationUnit.compile(CompilationUnit.java:574)
> 	at org.gradle.api.internal.tasks.compile.ApiGroovyCompiler.execute(ApiGroovyCompiler.java:268)
> 	at org.gradle.api.internal.tasks.compile.ApiGroovyCompiler.execute(ApiGroovyCompiler.java:68)
> 	at org.gradle.api.internal.tasks.compile.GroovyCompilerFactory$DaemonSideCompiler.execute(GroovyCompilerFactory.java:87)
> 	at org.gradle.api.internal.tasks.compile.GroovyCompilerFactory$DaemonSideCompiler.execute(GroovyCompilerFactory.java:75)
> 	at org.gradle.api.internal.tasks.compile.daemon.AbstractDaemonCompiler$CompilerWorkAction.execute(AbstractDaemonCompiler.java:113)
> 	at org.gradle.workers.internal.DefaultWorkerServer.execute(DefaultWorkerServer.java:47)
> 	at org.gradle.workers.internal.AbstractClassLoaderWorker$1.create(AbstractClassLoaderWorker.java:46)
> 	at org.gradle.workers.internal.AbstractClassLoaderWorker$1.create(AbstractClassLoaderWorker.java:36)
> 	at org.gradle.internal.classloader.ClassLoaderUtils.executeInClassloader(ClassLoaderUtils.java:98)
> 	at org.gradle.workers.internal.AbstractClassLoaderWorker.executeInClassLoader(AbstractClassLoaderWorker.java:36)
> 	at org.gradle.workers.internal.IsolatedClassloaderWorker.execute(IsolatedClassloaderWorker.java:54)
> 	at org.gradle.workers.internal.WorkerDaemonServer.execute(WorkerDaemonServer.java:56)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
> 	at org.gradle.process.internal.worker.request.WorkerAction.run(WorkerAction.java:118)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
> 	at org.gradle.internal.dispatch.ReflectionDispatch.dispatch(ReflectionDispatch.java:36)
> 	at org.gradle.internal.dispatch.ReflectionDispatch.dispatch(ReflectionDispatch.java:24)
> 	at org.gradle.internal.remote.internal.hub.MessageHubBackedObjectConnection$DispatchWrapper.dispatch(MessageHubBackedObjectConnection.java:182)
> 	at org.gradle.internal.remote.internal.hub.MessageHubBackedObjectConnection$DispatchWrapper.dispatch(MessageHubBackedObjectConnection.java:164)
> 	at org.gradle.internal.remote.internal.hub.MessageHub$Handler.run(MessageHub.java:412)
> 	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
> 	at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:48)
> 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
> 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
> 	at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:56)
> 	at java.base/java.lang.Thread.run(Thread.java:829)
> Caused by: java.lang.NoClassDefFoundError: com.google.protobuf.GeneratedMessageLite
> 	at org.codehaus.groovy.ast.decompiled.AsmReferenceResolver.resolveClass(AsmReferenceResolver.java:46)
> 	at org.codehaus.groovy.ast.decompiled.TypeSignatureParser.visitEnd(TypeSignatureParser.java:109)
> 	at groovyjarjarasm.asm.signature.SignatureReader.parseType(SignatureReader.java:206)
> 	at groovyjarjarasm.asm.signature.SignatureReader.accept(SignatureReader.java:122)
> 	at org.codehaus.groovy.ast.decompiled.ClassSignatureParser.parseClassSignature(ClassSignatureParser.java:74)
> 	at org.codehaus.groovy.ast.decompiled.ClassSignatureParser.configureClass(ClassSignatureParser.java:32)
> 	at org.codehaus.groovy.ast.decompiled.DecompiledClassNode.lazyInitSupers(DecompiledClassNode.java:183)
> 	at org.codehaus.groovy.ast.decompiled.DecompiledClassNode.getGenericsTypes(DecompiledClassNode.java:80)
> 	at org.codehaus.groovy.control.GenericsVisitor.checkGenericsUsage(GenericsVisitor.java:132)
> 	at org.codehaus.groovy.control.GenericsVisitor.checkGenericsUsage(GenericsVisitor.java:126)
> 	at org.codehaus.groovy.control.GenericsVisitor.visitDeclarationExpression(GenericsVisitor.java:105)
> 	at org.codehaus.groovy.ast.expr.DeclarationExpression.visit(DeclarationExpression.java:89)
> 	at org.codehaus.groovy.ast.CodeVisitorSupport.visitExpressionStatement(CodeVisitorSupport.java:120)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitExpressionStatement(ClassCodeVisitorSupport.java:197)
> 	at org.codehaus.groovy.ast.stmt.ExpressionStatement.visit(ExpressionStatement.java:40)
> 	at org.codehaus.groovy.ast.CodeVisitorSupport.visitBlockStatement(CodeVisitorSupport.java:86)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitBlockStatement(ClassCodeVisitorSupport.java:106)
> 	at org.codehaus.groovy.ast.stmt.BlockStatement.visit(BlockStatement.java:69)
> 	at org.codehaus.groovy.ast.CodeVisitorSupport.visitTryCatchFinally(CodeVisitorSupport.java:133)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitTryCatchFinally(ClassCodeVisitorSupport.java:232)
> 	at org.codehaus.groovy.ast.stmt.TryCatchStatement.visit(TryCatchStatement.java:42)
> 	at org.codehaus.groovy.ast.CodeVisitorSupport.visitBlockStatement(CodeVisitorSupport.java:86)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitBlockStatement(ClassCodeVisitorSupport.java:106)
> 	at org.codehaus.groovy.ast.stmt.BlockStatement.visit(BlockStatement.java:69)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitClassCodeContainer(ClassCodeVisitorSupport.java:110)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitConstructorOrMethod(ClassCodeVisitorSupport.java:121)
> 	at org.codehaus.groovy.ast.ClassCodeVisitorSupport.visitMethod(ClassCodeVisitorSupport.java:132)
> 	at org.codehaus.groovy.control.GenericsVisitor.visitMethod(GenericsVisitor.java:92)
> 	at org.codehaus.groovy.ast.ClassNode.visitContents(ClassNode.java:1103)
> 	at org.codehaus.groovy.control.GenericsVisitor.visitClass(GenericsVisitor.java:66)
> 	at org.codehaus.groovy.control.CompilationUnit$4.call(CompilationUnit.java:228)
> 	at org.codehaus.groovy.control.CompilationUnit.applyToPrimaryClassNodes(CompilationUnit.java:1084)
> 	... 36 more

I am not getting this error on any 0.x version. Is there a migration guide I need to follow? Please suggest

ankitatcodingnebula, 2022-07-11:
@JonasVautherin Thank you for the response back. I did but I still get the same error as above. I have this in my build.gradle

```
implementation("io.mavsdk:mavsdk:<versison>")  {
    exclude group: "com.google.protobuf", module: "protobuf-javalite"
}
```

JonasVautherin, 2022-07-11:
@divyanshu1234 would you have some idea by any chance? :pray: :innocent:

divyanshu1234, 2022-07-11:
This is weird. I had fixed the problem in the library itself. So it isn't supposed to come. @ankitatcodingnebula are you using any other proto based dependency in your project?

ankitatcodingnebula, 2022-07-12:
Hi @divyanshu1234 No other proto based dependency.

divyanshu1234, 2022-07-12:
Try to add the following as your dependencies:

```
def grpcVersion = "1.36.2"

implementation "io.grpc:grpc-protobuf:${grpcVersion}"
implementation "io.grpc:grpc-okhttp:${grpcVersion}"
implementation "io.grpc:grpc-protobuf-lite:${grpcVersion}"
implementation "io.grpc:grpc-stub:${grpcVersion}"
```

If this doesn't work, can you share your build.gradle file here?

ankitatcodingnebula, 2022-07-12:
@divyanshu1234 Adding grpc dependencies as mentioned above worked. Thanks. 

Could this be because these dependencies are added with `implementation` instead of `api` in the build.gradle of the artefact

build.gradle of mavsdk has this. Maybe changing them to api removes the need to re-declare these dependencies. Not sure.

```
compileOnly "io.grpc:grpc-protobuf:${grpcVersion}"
implementation "io.grpc:grpc-okhttp:${grpcVersion}"
implementation "io.grpc:grpc-protobuf-lite:${grpcVersion}"
implementation "io.grpc:grpc-stub:${grpcVersion}"
```

divyanshu1234, 2022-07-13:
@ankitatcodingnebula I had had the same doubt several months ago, but adding these dependencies through the `api` configuration will only cause conflicts when you add other protobuf dependencies to your project as they'll cause the duplicate class found error.

divyanshu1234, 2022-08-09:
@JonasVautherin. I think you can close this issue.

closed 2022-08-09 by julianoes