mavlink/MAVSDK-Java #101: How to stop manual control?

xumeng367, 2022-07-11:
When i start manual contorl ,i call  drone.getManualControl().startPositionControl(), but no stop method,so when stop call drone.getManualControl().setManualControlInputI(), the drone will lost control and return go home?

julianoes, 2022-07-11:
Yes, that's correct. If you no longer set the manual control input, it should detect it as a timeout and trigger the failsafe action configured for RC loss.

xumeng367, 2022-07-11:
How do I hover the drone and then execute a mission?

julianoes, 2022-07-11:
You send a takeoff using `action->takeoff()`, then you do `mission->start_mission()`. I would first try these things out in simulation and also using QGC to learn more about how it works.

closed 2022-07-11 by julianoes