mavlink/MAVSDK-Java #102: When take photo, subscribeCaptureInfo do not update?

xumeng367, 2022-07-18:
When i take photo, and Listen for this method
 Disposable captureInfoDisposable = drone.getCamera().getCaptureInfo().subscribe( captureInfo -> {
                    Log.d(TAG, "run: " + captureInfo);
                });

but i  unable to receive notification of picture info update, is this a Bug?

JonasVautherin, 2022-07-18:
Can you show your code?

xumeng367, 2022-07-18:
Yeah, take photo method:
                drone.getCamera().takePhoto().subscribe();

listener photo info when create:
Disposable captureInfoDisposable = drone.getCamera().getCaptureInfo().subscribe( captureInfo -> {
Log.d(TAG, "run: " + captureInfo);

when connect drone ,first take photo, It's ok ,getCaptureInfo is called ,but take photo again, no response, getCaptureInfo not called;disconnect the drone and reconnect , ,first take photo, It's ok again.but take photo again, no response, again

julianoes, 2022-07-18:
What are you testing against? Do you know if that implementation is actually sending the required message [CAMERA_IMAGE_CAPTURED](https://mavlink.io/en/messages/common.html#CAMERA_IMAGE_CAPTURED)?

JonasVautherin, 2022-07-18:
@julianoes just thinking: say the drone does not send the camera status, and hence does not increment `image_count`. Won't MAVSDK ignore new `CAMERA_IMAGE_CAPTURED` updates? :thinking: 

Next question would be: is it mandatory for a camera to send the camera status? :sweat_smile:

julianoes, 2022-07-21:
> is it mandatory for a camera to send the camera status?

Mandatory? Who knows, probably not. Does it work without it? Probably not :smile:

JonasVautherin, 2024-08-20:
This has been inactive for years, let's close it.

closed 2024-08-20 by JonasVautherin