mavlink/MAVSDK-Java #103: importQgroundcontrolMission(filePathString) does not recognise Android Filepath

ThomasCarstens, 2022-08-01:
I am building a mavsdk-enabled Android app with QGC running on my pc.
The .plan file is stored on the /data/ folder of my Android device.

The command rejects this path with a failure to parse the file. The file itself can be loaded and run with no problems on QGC...
Is it perhaps that this command cannot access non-system Android directories? I cannot find any documentation on this subject. 
Thanks for your time.

julianoes, 2022-08-02:
Do you have permissions in your app to read the files?

ThomasCarstens, 2022-08-02:
Yes, External Storage is configured in the Manifest file.
By the way, it does parse correctly with  "/home/me/AndroidProjects/myproject/main/src/assets/*.plan" (my PC subsystem). I also noticed that adb cannot read  my files in a app production build. Do you have a suggestion on how I should proceed?

ThomasCarstens, 2022-08-02:
Are you able to reproduce this issue? Thanks

julianoes, 2022-08-02:
> "/data/user/0/com.myapp.App/cache/path.plan"

Is this external storage? I guess not. But do you have access to the cache? I guess by default you would...

I'm sorry I'm not usually developing on Android, so if you want me to reproduce it, I'd need a minimal example that I could load and try out.

JonasVautherin, 2022-08-15:
Sounds like an Android permission issue (i.e. unrelated to MAVSDK itself). Are you sure your project is setup correctly and you app has access to this file on the device?

ThomasCarstens, 2022-08-15:
Hi JonasVautherin and julianoes, thanks for the tips. I suspect that a production build is causing issues, I should be able to read cache. We decided to try another approach, so I am not planning to look into this anymore. Thanks again!