mavlink/MAVSDK-Java #105: Parsing .plan string into MissionRaw with MissionRawException: Invalid Argument

ThomasCarstens, 2022-08-02:
Good day. I have a MAVSDK-Java enabled Android app and I am trying to add a MissionItem to my drone.

I create a mission item as such (this one has shown to fail)
```
List<MissionRaw.MissionItem> ArrayofMissionItems = new ArrayList<>();

MissionRaw.MissionItem item1 = new MissionRaw.MissionItem(
                int 1 #seq
                int 3 #frame, 
                int 84 #command,
                int 1 #current,
                int 1 #autocontinue,
                float 0.0 #param1,
                float 0.0 #param2,
                float 0.0 #param3,
                float 0.0 #param4,
                int 144444444 #x,
                int -177777777 #y,
                float 50.0 #z,
                int 0 #missionType
)
ArrayofMissionItems.add(item1)
drone.getMissionRaw()
                .uploadMission(ArrayofMissionItems).doOnComplete(() -> logger.info("Upload succeeded"))
                .doOnError(throwable -> logger.info("Issue uploading"))
                .subscribe();
```

Invalid Argument exception occurs on mission upload after following the MissionRaw argument layout from https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_mission_raw.html 
Are you able to see what is causing the issue? Thanks

ThomasCarstens, 2022-08-03:
Okay, it was just a matter of 50.0 replacing to 50f, it works now.

closed 2022-08-03 by ThomasCarstens

julianoes, 2022-08-03:
Oh wow, tricky! Thanks for the note.