mavlink/MAVSDK-Java #107: Is there any way to get Battery temperature?

xumeng367, 2022-08-17:
Hello, I have a question, if MAvSDK does not have the encapsulated data, is there a way to obtain the MAvlink data directly, like Battery temperature? How do I set the Home point for features that MAVSDK does not have, such as MAV_CMD_DO_SET_HOME?

julianoes, 2022-08-17:
There is currently no support for MAVLink passthrough in the language wrappers, so only in C++. It's one of the features I'm working on but progress is slow as I'm only working on that in my spare time, so to say, no one wanted to fund that work yet.

divyanshu1234, 2022-08-22:
We use a combination of MAVSDK (high-level operations) and raw MAVLink (low-level operations), to get complete control over the system.

You can have a look at [dronefleet/mavlink](https://github.com/dronefleet/mavlink) for Java or [urbanmatrix/mavlink-kotlin](https://github.com/urbanmatrix/mavlink-kotlin) (in development) for that.

The trick is to split the TCP/UDP/Serial MAVLink streams and provide them to the two libraries.

rkashaev, 2023-02-19:
Hello guys! 
I think I also need more detail batter info (like current consumption) as well as some other stuff. I also wanted to ask about `passthrough` thing, but @julianoes already said he's working on it. 
I think I really need this passthrough thing as a backdoor for a lot of missing stuff, so I want to suggest my help with Java transferring. If you guys help me with some contribution rules etc.
Also, I'm thinking about the same way @divyanshu1234 suggested. Please, can you give more details here?
As a desktop java application, can I connect directly to the mavlink device, or it's necessary to use mavsdk-server (as I use now)?
If I use the server, can I connect to the same flight-controller/server using both mavsdk-java and dronefleet/mavlink?

divyanshu1234, 2023-02-19:
@rkashaev if you're working on a desktop application, Then you can use mavlink-router to split the MAVLink input into two quite easily. I suggest going through the documentation for it.

For example, if you're using PX4 SITL then the mavlink-router command would look something like this:

```
mavlink-routerd 127.0.0.1:14550 -t 5760
```

This will connect to the UDP localhost 14550 source and create a TCP server at Port 5750.
You can now connect both mavsdk-server and mavlink-kotlin/dronefleet to it.

rkashaev, 2023-02-19:
Thanks, @divyanshu1234! Will try.. 
I have never ever touch any SITL, just my bare FC connected through a wire to my laptop and making a GCS on my laps literally..

divyanshu1234, 2023-02-19:
mavlink-router even supports serial connections. You can try:

```
mavlink-routerd /dev/ttyACM0:57600 -t 5760
```

Of course, the device and the baud rate may differ. Please go through the docs of mavlink-router. It's a great utility.

closed 2025-08-21 by JonasVautherin