mavlink/MAVSDK-Java #110: Not Able to connect android client app to PX4 sitl

Chandanrajshaw, 2022-08-30:
Hii!
I am trying  the android-client app. I m trying to make connection with PX4 sitl  for sending commands to the simulated jmavsim
for simulation but not able to connect the System(drone) to the mavsdk server running on android Phone  .
Logs: 
D/MAVSDK-Server: Running mavsdk_server with connection url: udp://:14540
I/Mavsdk: MAVSDK version: v0.37.0
D/Mavsdk: New: System ID: 0 Comp ID: 0
I/Mavsdk: Server started
W/k.androidclient: type=1400 audit(0.0:260764): avc: denied { read } for name="somaxconn" dev="proc" ino=7084178 scontext=u:r:untrusted_app:s0:c35,c257,c512,c768 tcontext=u:object_r:proc_net:s0 tclass=file permissive=0 app=io.mavsdk.androidclient
I/Mavsdk: Server set to listen on 0.0.0.0:40965
I/Mavsdk: Waiting to discover system on udp://:14540...
D/MAVSDK-Server: mavsdk_server is now running, listening on port 40965
V/Action: [main] - Building channel to 127.0.0.1:40965
V/Calibration: [main] - Building channel to 127.0.0.1:40965
V/Camera: [main] - Building channel to 127.0.0.1:40965
V/Core: [main] - Building channel to 127.0.0.1:40965
V/Failure: [main] - Building channel to 127.0.0.1:40965
V/FollowMe: [main] - Building channel to 127.0.0.1:40965
V/Ftp: [main] - Building channel to 127.0.0.1:40965
V/Geofence: [main] - Building channel to 127.0.0.1:40965
V/Gimbal: [main] - Building channel to 127.0.0.1:40965
V/Info: [main] - Building channel to 127.0.0.1:40965
V/LogFiles: [main] - Building channel to 127.0.0.1:40965
V/ManualControl: [main] - Building channel to 127.0.0.1:40965
V/Mission: [main] - Building channel to 127.0.0.1:40965
V/MissionRaw: [main] - Building channel to 127.0.0.1:40965
V/Mocap: [main] - Building channel to 127.0.0.1:40965
V/Offboard: [main] - Building channel to 127.0.0.1:40965
V/Param: [main] - Building channel to 127.0.0.1:40965
V/Shell: [main] - Building channel to 127.0.0.1:40965
V/Telemetry: [main] - Building channel to 127.0.0.1:40965
V/Tune: [main] - Building channel to 127.0.0.1:40965

Also I had tried to connect mavsdk server running on desktop from the android-client using IP address of the system where Mavsdk  server is running  with port :50051  but also it havent worked.
Logs :
V/Action: [main] - Building channel to 192.168.39.64:50051
V/Calibration: [main] - Building channel to 192.168.39.64:50051
V/Camera: [main] - Building channel to 192.168.39.64:50051
V/Core: [main] - Building channel to 192.168.39.64:50051
V/Failure: [main] - Building channel to 192.168.39.64:50051
V/FollowMe: [main] - Building channel to 192.168.39.64:50051
V/Ftp: [main] - Building channel to 192.168.39.64:50051
V/Geofence: [main] - Building channel to 192.168.39.64:50051
V/Gimbal: [main] - Building channel to 192.168.39.64:50051
V/Info: [main] - Building channel to 192.168.39.64:50051
V/LogFiles: [main] - Building channel to 192.168.39.64:50051
V/ManualControl: [main] - Building channel to 192.168.39.64:50051
V/Mission: [main] - Building channel to 192.168.39.64:50051
V/MissionRaw: [main] - Building channel to 192.168.39.64:50051
V/Mocap: [main] - Building channel to 192.168.39.64:50051
V/Offboard: [main] - Building channel to 192.168.39.64:50051
V/Param: [main] - Building channel to 192.168.39.64:50051
V/Shell: [main] - Building channel to 192.168.39.64:50051
V/Telemetry: [main] - Building channel to 192.168.39.64:50051
V/Tune: [main] - Building channel to 192.168.39.64:50051
I/System.out: io.mavsdk.System@9bb795a
W/Activity: Slow Operation: Activity io.mavsdk.androidclient/.MapsActivity onResume took 286ms
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/ConscryptEngineSocket;->setUseSessionTickets(Z)V (greylist-max-q,test-api, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->setUseSessionTickets(Z)V (greylist-max-q,test-api, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/AbstractConscryptSocket;->setUseSessionTickets(Z)V (greylist-max-q, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/ConscryptEngineSocket;->setHostname(Ljava/lang/String;)V (greylist-max-q,test-api, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->setHostname(Ljava/lang/String;)V (greylist-max-q,test-api, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/AbstractConscryptSocket;->setHostname(Ljava/lang/String;)V (greylist-max-q, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->setAlpnProtocols([B)V (greylist-max-q,test-api, reflection, denied)
W/k.androidclien: Accessing hidden method Lcom/android/org/conscrypt/AbstractConscryptSocket;->setAlpnProtocols([B)V (greylist-max-q, reflection, denied)
W/System.err: io.reactivex.exceptions.OnErrorNotImplementedException: UNAVAILABLE
W/System.err:     at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:704)
W/System.err:     at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:701)
W/System.err:     at io.reactivex.internal.subscribers.LambdaSubscriber.onError(LambdaSubscriber.java:79)
W/System.err:     at io.reactivex.internal.subscribers.BasicFuseableSubscriber.onError(BasicFuseableSubscriber.java:101)
W/System.err:     at io.reactivex.internal.util.NotificationLite.accept(NotificationLite.java:221)
W/System.err:     at io.reactivex.internal.operators.flowable.FlowableReplay$BoundedReplayBuffer.replay(FlowableReplay.java:860)
W/System.err:     at io.reactivex.internal.operators.flowable.FlowableReplay$ReplaySubscriber.onError(FlowableReplay.java:406)
W/System.err:     at io.reactivex.internal.operators.flowable.FlowableCreate$BaseEmitter.error(FlowableCreate.java:288)
W/System.err:     at io.reactivex.internal.operators.flowable.FlowableCreate$BufferAsyncEmitter.drain(FlowableCreate.java:531)
W/System.err:     at io.reactivex.internal.operators.flowable.FlowableCreate$BufferAsyncEmitter.tryOnError(FlowableCreate.java:481)
W/System.err:     at io.reactivex.internal.operators.flowable.FlowableCreate$BaseEmitter.onError(FlowableCreate.java:270)
W/System.err:     at io.mavsdk.telemetry.Telemetry$5.onError(Telemetry.java:1690)
W/System.err:     at io.grpc.stub.ClientCalls$StreamObserverToCallListenerAdapter.onClose(ClientCalls.java:487)
W/System.err:     at io.grpc.internal.ClientCallImpl.closeObserver(ClientCallImpl.java:562)
W/System.err:     at io.grpc.internal.ClientCallImpl.access$300(ClientCallImpl.java:70)
W/System.err:     at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1StreamClosed.runInternal(ClientCallImpl.java:743)
W/System.err:     at io.grpc.internal.ClientCallImpl$ClientStreamListenerImpl$1StreamClosed.runInContext(ClientCallImpl.java:722)
W/System.err:     at io.grpc.internal.ContextRunnable.run(ContextRunnable.java:37)
W/System.err:     at io.grpc.internal.SerializingExecutor.run(SerializingExecutor.java:133)
W/System.err:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
W/System.err:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
W/System.err:     at java.lang.Thread.run(Thread.java:923)
W/System.err: Caused by: io.grpc.StatusRuntimeException: UNAVAILABLE
W/System.err:     at io.grpc.Status.asRuntimeException(Status.java:535)
W/System.err: 	... 10 more
W/System.err: Caused by: java.net.ConnectException: failed to connect to /192.168.39.64 (port 50051) from /:: (port 46236): connect failed: ECONNREFUSED (Connection refused)
W/System.err:     at libcore.io.IoBridge.connect(IoBridge.java:142)
W/System.err:     at java.net.PlainSocketImpl.socketConnect(PlainSocketImpl.java:142)
W/System.err:     at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:390)
W/System.err:     at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:230)
W/System.err:     at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:212)
W/System.err:     at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:436)
W/System.err:     at java.net.Socket.connect(Socket.java:621)
W/System.err:     at java.net.Socket.connect(Socket.java:570)
W/System.err:     at java.net.Socket.<init>(Socket.java:450)
W/System.err:     at java.net.Socket.<init>(Socket.java:250)
W/System.err:     at javax.net.DefaultSocketFactory.createSocket(SocketFactory.java:285)
W/System.err:     at io.grpc.okhttp.OkHttpClientTransport$3.run(OkHttpClientTransport.java:529)
W/System.err: 	... 4 more
W/System.err: Caused by: android.system.ErrnoException: connect failed: ECONNREFUSED (Connection refused)

I am going through the documentation for development but not getting success. 
Plzzz Help me out.....Thanku.

JonasVautherin, 2022-08-30:
> I/Mavsdk: MAVSDK version: v0.37.0

Can you try an up-to-date version?

julianoes, 2022-08-31:
Does the app have permission to access the network?

Chandanrajshaw, 2022-08-31:
> Does the app have permission to access the network?

 <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE"/>
    <uses-permission android:name="android.permission.READ_PHONE_STATE" />
These permission I had Given In Manifest.xml

Chandanrajshaw, 2022-08-31:
> > I/Mavsdk: MAVSDK version: v0.37.0
> 
> Can you try an up-to-date version?

I had updated mavsdk & mavsdkserver version i.e.
  implementation 'io.mavsdk:mavsdk:1.2.0'
    implementation 'io.mavsdk:mavsdk-server:1.2.1'

After Updating only this is appearing in the log:
-08-31 13:27:34.663 20672-20731/io.mavsdk.androidclient I/Mavsdk: MAVSDK version: v1.4.3
2022-08-31 13:27:34.663 20672-20731/io.mavsdk.androidclient D/MAVSDK-Server: Running mavsdk_server with connection url: udp://:14540
2022-08-31 13:27:34.663 20672-20731/io.mavsdk.androidclient I/Mavsdk: Waiting to discover system on udp://:14540...

also code below 
"int   port =mavsdkServer.run();"
not Executing. Below some part of code Snipset I m including:
private void startMavsdkServer()
{

  MavsdkEventQueue.executor().execute(new Runnable() {
    @Override
    public void run() {
    int   port =mavsdkServer.run();
        handler.post(new Runnable() {
          @Override
          public void run() {
            Message msg=new Message();
              msg.obj=port;
            responseHandler.sendMessage(msg);
          }
        });
    }
  });
}
....................
....................
    drone = new System(BACKEND_IP_ADDRESS,port);
.................

JonasVautherin, 2022-08-31:
> "int port =mavsdkServer.run();" not Executing.

You mean it does not return? That's normal, it's blocking until the drone is discovered.

> Waiting to discover system on udp://:14540...

This says that MAVSDK is waiting for MAVLink messages to come on UDP port 14540. How are you pointing SITL to your Android app?

I could recommend trying this: if `192.168.1.12` is the IP of you Android device, try running SITL like this:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.0 192.168.1.12
```

This is using [px4-gazebo-headless](https://github.com/jonasvautherin/px4-gazebo-headless). Make sure you replace `192.168.1.12` with the IP of you Android device.

Chandanrajshaw, 2022-08-31:
Got Successfull in connecting to mav sdk server:
I/Mavsdk: MAVSDK version: v1.4.3
D/MAVSDK-Server: Running mavsdk_server with connection url: udp://:14540
I/Mavsdk: Waiting to discover system on udp://:14540...
I/Mavsdk: New system on: 192.168.39.64:62081 (with sysid: 1)
D/Mavsdk: New: System ID: 1 Comp ID: 1
D/Mavsdk: Component Autopilot (1) added.
D/Mavsdk: Discovered 1 component(s)
I/Mavsdk: System discovered
I/Mavsdk: Server started
W/mavsdk-event-qu: type=1400 audit(0.0:377383): avc: denied { read } for name="somaxconn" dev="proc" ino=12034775 scontext=u:r:untrusted_app:s0:c35,c257,c512,c768 tcontext=u:object_r:proc_net:s0 tclass=file permissive=0 app=io.mavsdk.androidclient
I/Mavsdk: Server set to listen on 0.0.0.0:37591
D/MAVSDK-Server: mavsdk_server is now running, listening on port 37591
V/Telemetry: [mavsdk-event-queue] - Building channel to 127.0.0.1:37591
But after this I m getting log as:
W/Mavsdk: sending again after 1.52403 s, retries to do: 1  (520).
W/Mavsdk: sending again, retries to do: 3  (CAL_GYRO0_ID).
D/MapsActivity: [grpc-default-executor-2] - armed: false
D/MapsActivity: [grpc-default-executor-2] - flight mode: HOLD
W/Mavsdk: sending again, retries to do: 2  (CAL_GYRO0_ID).
E/Mavsdk: Retrying failed (520)
W/Mavsdk: sending again, retries to do: 1  (CAL_GYRO0_ID).
E/Mavsdk: Error: Retrying failed get param busy timeout: CAL_GYRO0_ID
E/Mavsdk: Error: Param for gyro cal failed.
W/Mavsdk: sending again, retries to do: 3  (CAL_ACC0_ID).
W/Mavsdk: sending again, retries to do: 2  (CAL_ACC0_ID).
W/Mavsdk: sending again, retries to do: 1  (CAL_ACC0_ID).
E/Mavsdk: Error: Retrying failed get param busy timeout: CAL_ACC0_ID
E/Mavsdk: Error: Param for accel cal failed.
W/Mavsdk: sending again, retries to do: 3  (CAL_MAG0_ID).
W/Mavsdk: sending again, retries to do: 2  (CAL_MAG0_ID).
W/Mavsdk: sending again, retries to do: 1  (CAL_MAG0_ID).
E/Mavsdk: Error: Retrying failed get param busy timeout: CAL_MAG0_ID
E/Mavsdk: Error: Param for mag cal failed.
W/Mavsdk: sending again, retries to do: 3  (SYS_HITL).
W/Mavsdk: sending again, retries to do: 2  (SYS_HITL).
W/Mavsdk: sending again, retries to do: 1  (SYS_HITL).
E/Mavsdk: Error: Retrying failed get param busy timeout: SYS_HITL
E/Mavsdk: Error: Param to determine hitl failed.
W/Mavsdk: sending again, retries to do: 3  (CAL_GYRO0_ID).
W/Mavsdk: sending again, retries to do: 2  (CAL_GYRO0_ID).
W/Mavsdk: sending again, retries to do: 1  (CAL_GYRO0_ID).
E/Mavsdk: Error: Retrying failed get param busy timeout: CAL_GYRO0_ID
E/Mavsdk: Error: Param for gyro cal failed.

What is meant by this?
and When I m trying to take off it is still showing "No System"?
Why ? and what will be the correct next procedure?

JonasVautherin, 2022-08-31:
How did you run SITL? What command did you run exactly?

Chandanrajshaw, 2022-08-31:
> How did you run SITL? What command did you run exactly?

I am Using this command to start px4- gazebo headless and for pointing sitl to connect android as u quoted above

docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.0 192.168.1.12

JonasVautherin, 2022-08-31:
> 192.168.1.12

Is that the IP of your Android device? I wrote it as an example :thinking:

Chandanrajshaw, 2022-09-01:
> > 192.168.1.12
> 
> Is that the IP of your Android device? I wrote it as an example 🤔

Nope..ip:-25.249.12.203

JonasVautherin, 2022-09-01:
> 25.249.12.203

What kind of network is that? Is it a LAN? It feels like the MAVLink messages are going one direction (from your computer to your Android device) but not the other way round (from the Android device to your computer).

Chandanrajshaw, 2022-09-03:
> > 25.249.12.203
> 
> What kind of network is that? Is it a LAN? It feels like the MAVLink messages are going one direction (from your computer to your Android device) but not the other way round (from the Android device to your computer).

That's ri8 I was using mobile n/w that's why msg was going one Direction . Now its working fine.
One Problem is arising that when the System is not Discovered the App is crashing ..its Showing to Handle error. Can u plzzz help me out How can This problem will be resolved?

JonasVautherin, 2022-09-03:
> One Problem is arising that when the System is not Discovered the App is crashing

I can only guess since I have no idea about your code, but I would say that you should run `mavsdkServer.run()` in a background thread.

Chandanrajshaw, 2022-09-06:
I/Mavsdk: MAVSDK version: v1.4.3
D/MAVSDK-Server: Running mavsdk_server with connection url: udp://:14540
I/Mavsdk: Waiting to discover system on udp://:14540...
E/Mavsdk: bind error: Address already in use
E/Mavsdk: Connection failed: Bind error

Why is this happening ? 
I had Locked the phone and then unlocked and then this is happening when I m trying to re-connect... 
I think mavsdk server is not stopping.... or something else ?

JonasVautherin, 2022-09-06:
Did you call `mavsdkServer.stop()` and `mavsdkServer.destroy()` to stop it?

Chandanrajshaw, 2022-09-08:
> Did you call `mavsdkServer.stop()` and `mavsdkServer.destroy()` to stop it?

Destroy() i haven't included . Now its working Good.

Chandanrajshaw, 2022-09-08:
I want to Display Video  which is captured By camera .
Is it possible? If possible can u suggest some example.
Thanku

julianoes, 2022-09-08:
@Chandanrajshaw please don't hijack the thread with other questions.

closed 2022-09-08 by julianoes