mavlink/MAVSDK-Java #112: Is it possible to make FPV which shows the camera feed in Android ?

Chandanrajshaw, 2022-09-12:
Hii,
I am trying to make android application that can send commands to drone and show the camera feed directly to the smartphone while flying the drone  as like DJI FPV widget do. 
Is it possible ?
I had tried to search but No documentation or example found. please guide or Suggest 
thank you

JonasVautherin, 2022-09-12:
Yes it is possible. Did you have a look at the [examples](https://github.com/mavlink/MAVSDK-Java/tree/main/examples)?

Chandanrajshaw, 2022-09-12:
> Yes it is possible. Did you have a look at the [examples](https://github.com/mavlink/MAVSDK-Java/tree/main/examples)?

Yes i had gone through android-client example

divyanshu1234, 2022-09-12:
Some of the readily available libraries are [ExoPlayer](https://github.com/google/ExoPlayer). If you want better control over the stream, you can try to integrate the VLC Android library or the GStreamer.

Chandanrajshaw, 2022-09-14:
I m trying o get the camera component to display video feed using exoplayer.For this I had used the approach as:

camera=MavServer.drone.getCamera();
    camera.getVideoStreamInfo().subscribe(new FlowableSubscriber<Camera.VideoStreamInfo>() {
        @Override
        public void onSubscribe(Subscription s) {
        }
        @Override
        public void onNext(Camera.VideoStreamInfo videoStreamInfo) {
            videoUri=Uri.parse(videoStreamInfo.getSettings().getUri());
            Log.d("GCSActivity", "onNext: "+videoUri);
        }
        @Override
        public void onError(Throwable t) {
            Toast.makeText(GCSActivity.this, ""+t.getMessage(), Toast.LENGTH_SHORT).show();
        }
        @Override
        public void onComplete() {
        }
    }); 
But not getting anything on logs, Beside this below is the log i m getting :

E/libc: Access denied finding property "ro.vendor.df.effect.conflict"
E/libc: Access denied finding property "ro.vendor.knock.type"
W/k.androidclient: type=1400 audit(0.0:78002): avc: denied { read } for name="u:object_r:vendor_displayfeature_prop:s0" dev="tmpfs" ino=1366 scontext=u:r:untrusted_app:s0:c35,c257,c512,c768 tcontext=u:object_r:vendor_displayfeature_prop:s0 tclass=file permissive=0 app=io.mavsdk.androidclient
V/Camera: [mavsdk-event-queue] - Building channel to 127.0.0.1:40131
W/Looper: PerfMonitor looperActivity : package=io.mavsdk.androidclient/.GCSActivity time=0ms latency=351ms running=0ms  procState=2 ClientTransaction{ callbacks=[android.app.servertransaction.TopResumedActivityChangeItem] } historyMsgCount=2 (msgIndex=2 wall=346ms seq=1430 running=286ms runnable=9ms io=27ms late=4ms h=android.app.ActivityThread$H w=159)
D/SurfaceView: UPDATE null, mIsCastMode = false
W/Mavsdk: Received ack for not-existing command: 2504! Ignoring...
W/Mavsdk: Received ack for not-existing command: 2505! Ignoring...
W/Looper: PerfMonitor doFrame : time=58ms vsyncFrame=0 latency=349ms procState=2 historyMsgCount=3 (msgIndex=2 wall=346ms seq=1430 running=286ms runnable=9ms io=27ms late=4ms h=android.app.ActivityThread$H w=159)
W/Mavsdk: sending again after 0.502672 s, retries to do: 3  (2504).
W/Mavsdk: sending again after 0.503078 s, retries to do: 3  (2505).
W/Mavsdk: Received ack for not-existing command: 2504! Ignoring...
W/Mavsdk: Received ack for not-existing command: 2505! Ignoring...
W/Mavsdk: Dropping command 2504 that is already being sent
W/Mavsdk: Dropping command 2505 that is already being sent
W/Mavsdk: sending again after 1.0072 s, retries to do: 2  (2504).
W/Mavsdk: sending again after 1.00744 s, retries to do: 2  (2505).
W/Mavsdk: Received ack for not-existing command: 2504! Ignoring

### **I m Not using physical drone**
what is happening here i have no clue ...plzzz help me out..Thanku

Chandanrajshaw, 2022-09-19:
> > I m Not using physical drone
> 
> Are you using SITL? PX4 SITL does not send the `VIDEO_STREAM_INFORMATION` mavlink message. I'm afraid you should try with a real drone (or with a sim that does support the camera).

yes I m using SITL and launching as command mentioned below
### docker run --rm -it -p 8554:8554 jonasvautherin/px4-gazebo-headless:1.13.0  -v typhoon_h480 192.168.13.86
In the documentation it is mentioned that this supports video streaming.

and while trying to access the camera i m getting log as
```

V/Camera: [mavsdk-event-queue] - Building channel to 127.0.0.1:40131
D/SurfaceView: UPDATE null, mIsCastMode = false
W/Mavsdk: Received ack for not-existing command: 2504! Ignoring...
W/Mavsdk: sending again after 0.502672 s, retries to do: 3 (2504).
```
**What actually this means ?**
I need to know this...plzz clear this

JonasVautherin, 2022-09-19:
> In the documentation it is mentioned that this supports video streaming.

Yes, it does expose an RTSP video stream, but it is not implementing the MAVLink camera messages. So you can point e.g. VLC to that video stream, but MAVSDK won't give you the URL because SITL does not send those messages. It's just not implemented in SITL.

closed 2025-08-21 by JonasVautherin