mavlink/MAVSDK-Java #115: MavsdkServer crash for devices at API level loewr than 30

divyanshu1234, 2022-09-21:
Android crashes on app start with the following error:

```
java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "pthread_cond_clockwait" referenced by ...lib/arm64-v8a/libmavsdk_server.so
```

Upon doing some research, I found that `pthread_cond_clockwait` is only available for API 30+

@JonasVautherin any idea how to fix this?

JonasVautherin, 2022-09-21:
The C++ toolchain got updated to ndk r25b and api 33. It used to target api 23... Could it be that? :thinking:

divyanshu1234, 2022-09-21:
I've done some tests. Looks like it's happening from mavsdk-server version 1.2.2

So this is probably the reason.

divyanshu1234, 2022-09-21:
Can it be rolled back?

JonasVautherin, 2022-09-21:
Yes, I messed up with dockcross, I thought it did not matter. I guess updating the ndk version is fine, but not the api version. I honestly don't really know what it changes at the C++ level :thinking:. I'll try to rollback :+1:

JonasVautherin, 2022-09-21:
Can you access the tar archive [here](https://wetransfer.com/downloads/73faa65341ba74dfc2a418745cfea19720220921221746/e45dbee3ccbe0f6e547356ca54b5aea920220921221801/0ab9c5)? Not completely sure how to best share it. If you prefer I can send it to you over the PX4 Slack in a DM.

I built for android-arm64 using dockcross with api 23 (instead of 33). If that works, then I'll update dockcross and will make a release :+1:

divyanshu1234, 2022-09-22:
Looks like the transfer has expired. You can send it over Slack, no problem.

I'll also try to figure out how to do a Sonatype SNAPSHOT release. Don't know how you'll do it over MAVSDK C++ side though.

divyanshu1234, 2022-09-22:
Also, I feel that you should use NDK r25 instead of r25b because r25 is an LTS version. Also, r25b is the latest version and may be unstable.

divyanshu1234, 2022-09-22:
Wait. r25b is an LTS. Please ignore the last comment.

JonasVautherin, 2022-09-23:
Actually, aren't we seeing an issue related to [this](https://developer.android.com/about/versions/nougat/android-7.0-changes#ndk)? We build all our dependencies using the NDK :thinking:, but `pthread_cond_clockwait` comes from libc...

divyanshu1234, 2022-09-23:
Hmm... It could be. But I don't know how that worked. I just read that thing in a Stack overflow thread which confirmed with the fact that you had updated the min SDK version.

Just give me a few hours to test the new configuration.

JonasVautherin, 2022-09-23:
Yes, it's clear that keeping it to API 23 is solving our problem, but I am not sure if that is the "correct fix". It feels like something is wrong, but API 23 is fine with it. T

JonasVautherin, 2022-09-23:
Ok, I think you were right, and the answer is [here](https://developer.android.com/ndk/guides/common-problems#minsdkversion_set_higher_than_device_api_level):

> Problem: Your NDK API level is higher than the API supported by your device.
>
> Solution: Set your NDK API level (APP_PLATFORM) to the minimum version of Android your app supports.

I'll update dockcross right away.

divyanshu1234, 2022-09-24:
That's great. Thanks!

closed 2022-09-25 by divyanshu1234

JonasVautherin, 2022-09-26:
mavsdk-server:1.2.5 should be out on MavenCentral in a few hours, with the updated release :+1:.

divyanshu1234, 2022-09-26:
Awesome!