mavlink/MAVSDK-Java #118: Execution failed for task ':sdk:mavsdk:generateProto'

alariq, 2022-10-06:
I have the next problem:

`Execution failed for task ':sdk:mavsdk:generateProto'.
> protoc: stdout: . stderr: Traceback (most recent call last):
    File "C:\Users\alari\AppData\Local\Programs\Python\Python310\Scripts\protoc-gen-mavsdk-script.py", line 33, in <module>
      sys.exit(load_entry_point('protoc-gen-mavsdk', 'console_scripts', 'protoc-gen-mavsdk')())
    File "c:\users\alari\androidstudioprojects\sdk\proto\pb_plugins\protoc_gen_mavsdk\__main__.py", line 17, in main
      AutoGen.generate_reactive(request).SerializeToString())
    File "c:\users\alari\androidstudioprojects\sdk\proto\pb_plugins\protoc_gen_mavsdk\autogen.py", line 40, in generate_reactive
      enums = Enum.collect_enums(plugin_name,
    File "c:\users\alari\androidstudioprojects\sdk\proto\pb_plugins\protoc_gen_mavsdk\enum.py", line 61, in collect_enums
      _enums[enum.name] = Enum(plugin_name,
    File "c:\users\alari\androidstudioprojects\sdk\proto\pb_plugins\protoc_gen_mavsdk\enum.py", line 18, in __init__
      self._template = template_env.get_template("enum.j2")
    File "C:\Users\alari\AppData\Local\Programs\Python\Python310\lib\site-packages\jinja2\environment.py", line 1010, in get_template
      return self._load_template(name, globals)
    File "C:\Users\alari\AppData\Local\Programs\Python\Python310\lib\site-packages\jinja2\environment.py", line 969, in _load_template
      template = self.loader.load(self, name, self.make_globals(globals))
    File "C:\Users\alari\AppData\Local\Programs\Python\Python310\lib\site-packages\jinja2\loaders.py", line 126, in load
      source, filename, uptodate = self.get_source(environment, name)
    File "C:\Users\alari\AppData\Local\Programs\Python\Python310\lib\site-packages\jinja2\loaders.py", line 220, in get_source
      raise TemplateNotFound(template)
  jinja2.exceptions.TemplateNotFound: enum.j2
  --mavsdk_out: protoc-gen-mavsdk: Plugin failed with status code 1.`

I did some investigation and found out that if I change 
`option "template_path=$project.rootDir/templates/"`
to 
`option "template_path=./templates/"` (relative path) 
it works ok (well there is another [(issue)](https://github.com/mavlink/MAVSDK-Java/issues/117) but ther eis a workaround.

However, when I add this project to my app by using: `includeBuild '../sdk'` I receive same issue.
It looks like problem is partially related to gradle version, because in this case I alaso have issue with checkstyle task, it complaines about :
` configDir = project.provider({ rootProject.file("config/checkstyle") })`
so I looked up that configDir now has to be changed to configDirectory.
When I do this change I get to the firsdt error, related to template folder.

Any ideas on how can I make it work in my project provided that it compiles ok separately in console (after changing template folder path to relative form)

Thank you!

julianoes, 2022-10-09:
Thanks for the note @alariq. I'm hoping @JonasVautherin can help here.

JonasVautherin, 2022-10-09:
Which version of gradle are you using? Are you not using the gradle wrapper provided with the project?

I am not using it on Windows, is it different? I would expect Gradle to behave similarly, but maybe there are tricks with the paths there?

alariq, 2022-10-10:
It could be indeed that path issue is platform dependent. I also suspect that second issue is path related as wel as when I add some debug code to python, I see only first letter of disk, probably something is going on with ':' separator.

JonasVautherin, 2022-10-11:
Would be nice if you could investigate a bit, it really feels like this is related to the paths on Windows. Would be nice to understand if that's supposed to work like on Linux or not :+1:.

alariq, 2022-10-15:
Hi, I am a bit under pressure now, if I have some time in the future I'll have a look

closed 2025-08-21 by JonasVautherin