mavlink/MAVSDK-Java #120: Getting INVALID_ARGUMENT while uploading an ArduCopter mission

divyanshu1234, 2023-01-06:
MAVSDK always throws an `INVALID_ARGUMENT` error while uploading an ArduCopter mission. At the same time, QGC is able to upload the exact same mission.

Setup:
- ArduCopter SITL running on my local machine
- `mavlink-router` (with logger in between) to route messages from SITL to QCG on the local machine and my phone on the same network

I have gone down to replicating every param of the `MissionItemInt` generated in QGC using the `MissionRaw.MissionItemInt` of MAVSDK. I have noticed that MAVSDK throws the error before even sending a single message to the drone.

At the same time, the geofence upload function is working perfectly fine for the same ArduCopter.

divyanshu1234, 2023-01-06:
As an example, this is a sequence of mission items generated using QGC

```
MissionItemInt{targetSystem=1, targetComponent=1, seq=0, frame=MAV_FRAME_GLOBAL, command=MAV_CMD_NAV_WAYPOINT, current=1, autocontinue=1, param1=0.0, param2=0.0, param3=0.0, param4=0.0, x=-353632622, y=1491652375, z=0.0, missionType=MAV_MISSION_TYPE_MISSION}
MissionItemInt{targetSystem=1, targetComponent=1, seq=1, frame=MAV_FRAME_GLOBAL_RELATIVE_ALT, command=MAV_CMD_NAV_TAKEOFF, current=0, autocontinue=1, param1=0.0, param2=0.0, param3=0.0, param4=0.0, x=-353632622, y=1491652375, z=50.0, missionType=MAV_MISSION_TYPE_MISSION}
MissionItemInt{targetSystem=1, targetComponent=1, seq=2, frame=MAV_FRAME_GLOBAL, command=MAV_CMD_NAV_RETURN_TO_LAUNCH, current=0, autocontinue=1, param1=0.0, param2=0.0, param3=0.0, param4=0.0, x=0, y=0, z=0.0, missionType=MAV_MISSION_TYPE_MISSION}
```

It is being uploaded quite easily, while the exact sequence of MAVSDK messages throws the error.

JonasVautherin, 2023-01-06:
> I have noticed that MAVSDK throws the error before even sending a single message to the drone.

That's interesting :thinking:.

Having a quick look at the code, I think it probably comes from [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission/mission_impl.cpp#L1179-L1180), which maybe (?) comes from [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission/mission_impl.cpp#L163), but then it seems to come from a [MISSION_ACK here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/mavlink_mission_transfer.cpp#L521)?

Not sure... do you have a way to debug that in the C++ code directly? See what happens more precisely...

divyanshu1234, 2023-01-07:
Yes, even I was going through the C++ code yesterday and couldn't find why this was happening. As I had said, I couldn't see any `MISSION_ITEM_INT` emitted by MAVSDK, and therefore any `MISSION_ACK` from the drone's side.

I'll try to do some debugging on the C++ side today and figure out.

divyanshu1234, 2023-01-07:
Somewhat figured it out. It was a problem related to the sequencing of mission items [over here](https://github.com/mavlink/MAVSDK/blob/2a3032ed72422870c571de89fa952610d4b6e20f/src/mavsdk/core/mavlink_mission_transfer.cpp#L248)

It became difficult to debug it as the mavsdk_server version used in MAVSDK-Java probably threw `InvalidArgument` instead of `InvalidSequence` (in the latest version).

closed 2023-01-07 by divyanshu1234

julianoes, 2023-01-09:
Yes, sorry this will improve in the future.