mavlink/MAVSDK-Java #121: maven release?

rkashaev, 2023-01-11:
Hello, guys! Thanks for your work!

Not sure if it's the proper place to ask, but anyway. I'm trying to update the mavsdk-java lib to the latest release, but in the maven repo https://mvnrepository.com/artifact/io.mavsdk/mavsdk I see the latest version is 1.2.0 only.
Are there any plans for updating it? 
Currently, I'm struggling with building the lib from the sources and it's not that easy.

Thanks again!
Happy new year!

JonasVautherin, 2023-01-13:
What do you need that is not released yet?

I'll try to make a release early next week :+1:

rkashaev, 2023-01-15:
Oh, actually I confused with MAVSDK release. So yeah, according to the sources, latest mavsdk-java is really 1.2.0 (supporting server version 1.4.6), while MAVSDK itself is 1.4.10.

JonasVautherin, 2023-01-15:
So you don't really need a new release, do you? :thinking:

rkashaev, 2023-01-16:
I do! If it's possible, I would like to have MAVSDK-Java reflecting the latest MAVSDK version.

JonasVautherin, 2023-01-16:
I'll do it ASAP. Note that those are only a few bugfixes, so possibly it doesn't make a difference for you :+1:

divyanshu1234, 2023-01-21:
I've been trying to add more Adrupilot support but it looks like some changes have been done post 1.4.6

@JonasVautherin can you please do it by next week?

JonasVautherin, 2023-01-22:
Done! MAVSDK-Java 1.3.0 should be available on the Maven Repository in a few hours

divyanshu1234, 2023-01-23:
Thanks @JonasVautherin. I'll test it and get back in a few hours.

rkashaev, 2023-02-19:
Thanks, guys! I think we can close this!

closed 2023-02-19 by rkashaev