mavlink/MAVSDK-Java #125: MAVSDK Logging

jreedy-riis, 2023-03-21:
Hello! I was looking through the C++ documentation for logging with MAVSDK and saw that there was [a function to re-route logs from `stdout` to a callback](https://mavsdk.mavlink.io/main/en/cpp/guide/logging.html). Is this possible with MAVSDK-Java? I was looking through the plugins and was only able to view the log files themselves. If not, is it possible to use MAVSDK C++ and MAVSDK-Java in tandem where I can use MAVSDK C++ just to get log information and the java version for everything else? Or do each of these SDKs need to be initialized with the drone individually and therefore only one can be used?

JonasVautherin, 2023-03-21:
No this API is not available in Java. But I would have thought that `MavsdkServer` would print some logs. Doesn't it?

jreedy-riis, 2023-03-21:
It does print some logs, however, I would like to be able to show them on the UI on Android.

JonasVautherin, 2023-03-21:
Right. We don't have that now, not sure how I would do it to be honest.

julianoes, 2023-03-22:
@jreedy-riis do you need the actual mavsdk warnings and errors (which are not supported) or the MAVLink statustext warnings/errors sent by PX4? The latter can be accessed via telemetry.statustext.

jreedy-riis, 2023-03-22:
@julianoes That looks to be exactly what I need, thank you! I did see status text before I guess I just didn't see the documentation describing it. Thank you both for responding to my issue!

closed 2023-03-22 by jreedy-riis