mavlink/MAVSDK-Java #126: android-client example not connecting to simulated drone and crashing when pressing Takeoff or Land button

iura1, 2023-03-26:
Hi all, Does anyone have experience running the MAVSDK-Java android-client example? I can run it on the emulator in Android Studio but it is not connecting to Jmavsim Simulator, and it is crashing when I press land or takeoff button. I am using mavsdk_server built from sources. Please check the attached screenshots. Thx!

[image]
[image]

JonasVautherin, 2023-03-26:
Please link the other posts when you post the same question on different places: https://discuss.px4.io/t/android-client-example-not-connecting-to-simulated-drone-and-crashing-when-pressing-takeoff-or-land-button/31383

What I answered there: 

> I think you have to check the networking there. Your emulator is (I believe) not on the host network (it has an IP like 10.0.2.2, something like that). So you’ll have to send mavlink from jmavsim to the emulated device somehow.

julianoes, 2023-03-27:
Follow up over there, closing this.

closed 2023-03-27 by julianoes