mavlink/MAVSDK-Java #127: Camera Subscription Issue

jreedy-riis, 2023-04-07:
# Description

Hello! I have been trying to subscribe to the drone's `camera.status` and `camera.information`; however, the subscriptions never succeed. I can see the following logs from `MAVSDK`. It looks like the request and acknowledgement are successfully being sent, but MAVSDK thinks the `ack` is acknowledging an unknown command. When trying to do this using the simulator, I get no logs like below, or any values when using the `typhoon_h480`. However, when connected to a VOXL m500, I see the logs. Does this seem to be more of a drone issue or a MAVSDK issue? Thank you!

# Logs
```
sending again after 1.50492 s, retries to do: 1  (527).
sending again after 1.50512 s, retries to do: 1  (525).
Received ack for not-existing command: 527! Ignoring...
Received ack for not-existing command: 525! Ignoring...
Retrying failed (527)
Retrying failed (525)
```

# Code
```kotlin
camera?.status?.zipWith(camera?.information) { status, information ->            
            CameraInformation(
                availableStorage = status.availableStorageMib,
                usedStorage = status.usedStorageMib,
                totalStorage = status.totalStorageMib,
                verticalResolution = information.verticalResolutionPx,
                horizontalResolution = information.horizontalResolutionPx
            )
        }?.subscribe(
            { data -> emitCallback(data) },
            { e -> Log.d(TAG, "callback:  $e") }
        )
```

JonasVautherin, 2023-04-07:
SITL does not have a camera module. It just sends a video stream, but it does not support the MAVLink camera messages.

jreedy-riis, 2023-04-08:
Thank you for responding so quickly! Are you saying that the simulator doesn't support the MAVLink camera messages or MAVSDK-Java? It is fine that the simulator doesn't but I was experiencing the above error logs / no camera information with the VOXL m500.

JonasVautherin, 2023-04-08:
MAVSDK definitely supports cameras, it is used in production by different apps (including ESRI SiteScan).

I thought that the simulator did not, but [apparently](https://github.com/mavlink/MAVSDK-Python/issues/572#issuecomment-1500750678) with some models in the simulator there is some camera support. Maybe worth looking into that :wink:.

jreedy-riis, 2023-04-17:
Okay thank you!