mavlink/MAVSDK-Java #129: How to use MAVSDK-Java

JoshikaNetha, 2023-06-08:
Hi,
"I'm new to Mavsdk-Java , and I've successfully used Mavsdk in Python. However, I'm now interested in using it in Java. Could someone kindly assist me in utilizing Mavsdk in Java?

I would like to integrate Mavsdk-Java into my Java web application. Can anyone provide suggestions on how to integrate the necessary dependencies and incorporate Mavsdk-Java within a Java web application.

Thanks in advance.

JoshikaNetha, 2023-06-09:
Yes, I have seen READMe but I tried it , I have downloaded libmavsdk, I'm understanding how to run MAVSDK, So, please help me with this issue.

Thanks in advance

JoshikaNetha, 2023-06-12:
Thanks for response.

As you said I tried to build mavsdk-server from source file and from release, but its showing some error it not building successfully.

After running below command:

`cmake -Bbuild/default -DMAVLINK_DIALECT=mydialect -DMAVLINK_HEADERS=../mavlink-headers -H.`

Error is:
```

-- Version: v1.4.0-490-ged7ac077
-- Version major: 1
-- Version minor: 4
-- Version patch: 0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nsl/MavSDK/Mavsdk server/MAVSDK/build/default/third_party/mavlink
[ 12%] Performing update step for 'mavlink'
[ 25%] Performing configure step for 'mavlink'
-- MAVLink dialect: mydialect
-- MAVLink version: 2.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nsl/MavSDK/Mavsdk server/MAVSDK/build/default/third_party/mavlink/mavlink/src/mavlink-build
[ 37%] Performing build step for 'mavlink'
make[5]: *** No rule to make target '/home/nsl/MavSDK/Mavsdk server/MAVSDK/build/default/third_party/mavlink/mavlink/src/mavlink/message_definitions/v1.0/mydialect.xml', needed by 'include/mavlink/mydialect/mavlink.h'.  Stop.
make[4]: *** [CMakeFiles/Makefile2:76: CMakeFiles/generate_c_headers.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/mavlink.dir/build.make:112: mavlink/src/mavlink-stamp/mavlink-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/mavlink.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  /home/nsl/MavSDK/Mavsdk server/MAVSDK/build/default/third_party/mavlink
  failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:11 (build_target)

-- Configuring incomplete, errors occurred!
See also "/home/nsl/MavSDK/Mavsdk server/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".
See also "/home/nsl/MavSDK/Mavsdk server/MAVSDK/build/default/CMakeFiles/CMakeError.log".
```
I'm follow installation steps from below link:

[(https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html)]

I have followed **Building MAVSDK Library from Source steps** also. but not working .

Please help me with this issue.

divyanshupundir, 2023-06-12:
@JoshikaNetha No need to build the mavsdk-server. You can simply donwload the binary from the assets on the release page.

For example, if you have a Windows machine, you can download [`mavsdk-windows-x64-release.zip`](https://github.com/mavlink/MAVSDK/releases/download/v1.4.16/mavsdk-windows-x64-release.zip)

[image]

JoshikaNetha, 2023-06-12:
I have downloaded mavsdk-binary file as I'm working on ubuntu I have downloaded .deb file, then I have gave execution permission.

```
./libmavsdk-dev_1.4.16_ubuntu20.04_amd64.deb udp://:8240 -p 9000
./libmavsdk-dev_1.4.16_ubuntu20.04_amd64.deb: line 1: syntax error near unexpected token `newline'
./libmavsdk-dev_1.4.16_ubuntu20.04_amd64.deb: line 1: `!<arch>'
```

I'm getting above error.

divyanshupundir, 2023-06-12:
Please try [mavsdk_server_manylinux2010-x64](https://github.com/mavlink/MAVSDK/releases/download/v1.4.16/mavsdk_server_manylinux2010-x64).

JoshikaNetha, 2023-06-12:
Thanks so much It running successfully but, I'm running make px4_sitl  gazebo in same system.

In my Java Code I have written  as below:

		```
System system = new System("127.0.0.1", 9000);
		system.getAction().takeoff();
```

I have added all the dependencies in pom.xml as below:
```
  <dependencies>
  	<dependency>
  		<groupId>io.mavsdk</groupId>
  		<artifactId>mavsdk</artifactId>
  		<version>1.3.1</version>
  	</dependency>
  	<dependency>
  		<groupId>io.reactivex.rxjava2</groupId>
  		<artifactId>rxjava</artifactId>
  		<version>2.2.21</version>
  	</dependency>

  	<dependency>
  		<groupId>io.grpc</groupId>
  		<artifactId>grpc-okhttp</artifactId>
  		<version>1.3.1</version>
  		<scope>runtime</scope>
  	</dependency>
  	<dependency>
  		<groupId>io.grpc</groupId>
  		<artifactId>grpc-stub</artifactId>
  		<version>1.42.1</version>
  		<scope>runtime</scope>
  	</dependency>
  	<dependency>
  		<groupId>org.slf4j</groupId>
  		<artifactId>slf4j-api</artifactId>
  		<version>1.7.35</version>
  		<scope>runtime</scope>
  	</dependency>
  	<dependency>
  		<groupId>io.grpc</groupId>
  		<artifactId>grpc-protobuf-lite</artifactId>
  		<version>1.42.1</version>
  		<scope>runtime</scope>
  	</dependency>
  </dependencies>
```

Here, I'm running Maven Project as Maven build , I'm getting below error:

[```
[INFO] Scanning for projects...
[INFO] 
[INFO]  [1m--------------------------< [0;36mcom.maven:MavSDK[0;1m >--------------------------[m
[INFO]  [1mBuilding MavSDK 0.0.1-SNAPSHOT[m
[INFO]  [1m--------------------------------[ jar ]---------------------------------[m
[INFO] [1m------------------------------------------------------------------------[m
[INFO] BUILD FAILURE[m
[INFO] -----------------------------------------------------------------------[m
[INFO] Total time:  0.306 s
[INFO] Finished at: 2023-06-12T17:10:14+05:30
INFO] [1m------------------------------------------------------------------------[m
[ERROR] Unknown lifecycle phase "Run". You must specify a valid lifecycle phase or a goal in the format <plugin-prefix>:<goal> or <plugin-group-id>:<plugin-artifact-id>[:<plugin-version>]:<goal>. Available lifecycle phases are: validate, initialize, generate-sources, process-sources, generate-resources, process-resources, compile, process-classes, generate-test-sources, process-test-sources, generate-test-resources, process-test-resources, test-compile, process-test-classes, test, prepare-package, package, pre-integration-test, integration-test, post-integration-test, verify, install, deploy, pre-clean, clean, post-clean, pre-site, site, post-site, site-deploy. -> [1m[Help 1][m

[ERROR] To see the full stack trace of the errors, re-run Maven with the [1m-e[m switch.
[ERROR] Re-run Maven using the [1m-X[m switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [1m[Help 1][m http://cwiki.apache.org/confluence/display/MAVEN/LifecyclePhaseNotFoundException

```

divyanshupundir, 2023-06-12:
Try running the [TakeoffAndLand.java](https://github.com/mavlink/MAVSDK-Java/blob/main/examples/java-client/src/main/java/io/mavsdk/example/TakeoffAndLand.java) example with the correct address and port.

divyanshupundir, 2023-06-12:
Also, you only need to add the mavsdk dependency. No need to add the grpc and rxjava ones.

JoshikaNetha, 2023-06-12:
I have  mavsdk dependency also

```
<dependencies>
  	<dependency>
  		<groupId>io.mavsdk</groupId>
  		<artifactId>mavsdk</artifactId>
  		<version>1.3.1</version>
  	</dependency>
```

JoshikaNetha, 2023-06-12:
I tried to run TakeOffAndLand.java code I have mentioned 
         
         `System drone = new System("127.0.0.1", 9000); 
`
 using below commands:

```
:~/MavSDK/MAVSDK-Java/examples/java-client/src/main/java/io/mavsdk/example$ javac TakeoffAndLand.java 
TakeoffAndLand.java:3: error: cannot find symbol
import io.mavsdk.System;
                ^
  symbol:   class System
  location: package io.mavsdk
TakeoffAndLand.java:4: error: package io.mavsdk.action does not exist
import io.mavsdk.action.Action;
                       ^
TakeoffAndLand.java:7: error: package org.slf4j does not exist
import org.slf4j.Logger;
                ^
TakeoffAndLand.java:8: error: package org.slf4j does not exist
import org.slf4j.LoggerFactory;
                ^
TakeoffAndLand.java:11: error: cannot find symbol
  private static final Logger logger = LoggerFactory.getLogger(TakeoffAndLand.class);
                       ^
  symbol:   class Logger
  location: class TakeoffAndLand
TakeoffAndLand.java:11: error: cannot find symbol
  private static final Logger logger = LoggerFactory.getLogger(TakeoffAndLand.class);
                                       ^
  symbol:   variable LoggerFactory
  location: class TakeoffAndLand
TakeoffAndLand.java:16: error: constructor System in class System cannot be applied to given types;
    System drone = new System("127.0.0.1", 9000);
                   ^
  required: no arguments
  found:    String,int
  reason: actual and formal argument lists differ in length
TakeoffAndLand.java:19: error: cannot find symbol
    drone.getAction().arm()
         ^
  symbol:   method getAction()
  location: variable drone of type System
TakeoffAndLand.java:22: error: package Action does not exist
                  + ((Action.ActionException) throwable).getCode()))
                            ^
TakeoffAndLand.java:23: error: cannot find symbol
          .andThen(drone.getAction().takeoff()
                        ^
  symbol:   method getAction()
  location: variable drone of type System
TakeoffAndLand.java:26: error: package Action does not exist
                    + ((Action.ActionException) throwable).getCode())))
                              ^
TakeoffAndLand.java:28: error: cannot find symbol
          .andThen(drone.getAction().land()
                        ^
  symbol:   method getAction()
  location: variable drone of type System
TakeoffAndLand.java:31: error: package Action does not exist
                    + ((Action.ActionException) throwable).getCode())))
                              ^
13 errors

```

divyanshupundir, 2023-06-13:
Looks like a build problem. Are you not using a build system like Gradle? For big projects with several dependencies, we should try not to use `javac`, instead use Gradle or Idea build systems.

If you have IntelliJ installed on your system, you can simply open this project and run the TakeoffAndLand test.

sannithrao2468, 2023-06-14:
Hi ,@divyanshupundir . 
I have followed all the above steps but i'm stuck at the build step. Here is my error

madhavanenisannith@nhhydl-00271:~/Desktop/Workspace/MAVSDK-Java/examples/java-client$ ./gradlew clean build
> Task :compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Could not load class 'com.sun.tools.javac.api.JavacTool

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 397ms
2 actionable tasks: 2 executed

Previously i got an other error "cannot find tools.jar" , i resolved it , then i got this error. Can you please help?

divyanshupundir, 2023-06-14:
This is more of a problem with the project and Java setup than MAVSDK. It looks like Gradle isn't able to find that Java compiler on your machine.

An easy way you can fix it is by opening the project in IntelliJ Idea and it will install all the required dependencies. You can then even run the tests directly from it.

Otherwise, you can follow [this](https://stackoverflow.com/questions/20589718/compiler-error-error-could-not-find-or-load-main-class-com-sun-tools-javac-m) StackOverflow answer.