mavlink/MAVSDK-Java #130: Set Guided mode with MAVSDK-java

TeoVal138, 2023-06-09:
Hello everyone, 
I'm developing an android mobile app with mavsdk-java to control my ardupilot drone, in this days I've been trying to set the guided mode in order to takeoff autonomously.
Can you please tell me what can I do?

Many thanks in advance

JonasVautherin, 2023-06-09:
I think that the equivalent to "guided" in PX4 is "offboard". Did you try the offboard plugin? Not sure if it works with Ardupilot though :thinking:.

JonasVautherin, 2023-06-09:
Calling `offboard.start()` should just change the flight mode, see [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/offboard/offboard_impl.cpp#L56):

```c++
_system_impl->set_flight_mode(FlightMode::Offboard)
```

which ends up translating that to an Ardupilot flight mode [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#LL839C9-L839C71):

```c++
make_command_ardupilot_mode(flight_mode, component_id);
```

which e.g. for a copter will translate to `Guided` [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L964-L965):

```c++
        case FlightMode::Offboard:
            return ardupilot::CopterMode::Guided;
```

So I think it should work.

The problem is that at least the last part is not available in MAVSDK v1.4.16 (latest release, see [here](https://github.com/mavlink/MAVSDK/blob/v1.4.16/src/mavsdk/core/system_impl.cpp#L962)), only on the `main` branch.

I see two solutions for you (none of them ideal, unfortunately):
1. You build MAVSDK from source (from the C++ core up to MAVSDK-Java)
2. You wait until MAVSDK v2 is out, but that may take time. It is being tracked [here](https://github.com/mavlink/MAVSDK/issues/2002).

TeoVal138, 2023-06-09:
Thank you Jonas I will try to follow your advice! 😊
So no other mavsdk-java plugin would allow me to switch modes right?
I'm not just talking about the guided mode set but also about the other modes set

closed 2023-06-14 by TeoVal138