mavlink/MAVSDK-Java #131: MavSDK crashes on MavsdkServer initialization

IlfiDev, 2023-06-19:
Here's the error: 
`java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZNSt6__ndk14__fs10filesystem21__temp_directory_pathEPNS_10error_codeE" referenced by "/data/app/~~1nuV_Xwwx0jCemUqnF9G5w==/com.autel.maxlink-SBJ-pqkn17RYnGg-2j_YGA==/lib/arm64/libmavsdk_server.so"...`
This does not happen on older versions. 
What could be the reason and is there any fixes for that?

JonasVautherin, 2023-06-19:
I wonder if it could be related to https://github.com/mavlink/MAVSDK/pull/2005/files :thinking:

Maybe `${CMAKE_DL_LIBS}` should be added [after this line](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/CMakeLists.txt#L37)?

Is there a way you could try that (i.e. build mavsdk_server from sources with the change above) and let me know if it helps?

IlfiDev, 2023-06-29:
Addind `${CMAKE_DL_LIBS}` and bulidng did fix my problem. Thanks!