mavlink/MAVSDK-Java #132: Uploading Mission

TeoVal138, 2023-06-19:
Hi, I'm trying to upload a mission to a SITL drone connected to my android gcs made with mavsdk-java, but i always get this error:

**D/LOG_DroneRepository: About to upload 4 mission items
D/LOG_DroneRepository: Uploading mission...
D/LOG_DroneRepository: clearMission
D/LOG_DroneRepository: Upload setReturnToLaunchAfterMission
I/art: Do partial code cache collection, code=48KB, data=53KB
I/art: After code cache collection, code=48KB, data=53KB
I/art: Increasing code cache capacity to 256KB
I/art: Compiler allocated 8MB to compile void com.google.protobuf.MessageSchema.mergeFromHelper(com.google.protobuf.UnknownFieldSchema, com.google.protobuf.ExtensionSchema, java.lang.Object, com.google.protobuf.Reader, com.google.protobuf.ExtensionRegistryLite)
D/Mavsdk: current: 0, total: 4
D/LOG_DroneRepository: Failed to upload the mission INVALID_ARGUMENT: Invalid Argument**

this is the relative code:
        List<Mission.MissionItem> missionItems = new ArrayList<>();
        missionItems.add(generateMissionItem(-35.36235859999997, 149.1642425400000002));
        missionItems.add(generateMissionItem(-35.36235859999997, 149.1643425400000002));
        missionItems.add(generateMissionItem(-35.36235859999997, 149.1644425400000002));
        missionItems.add(generateMissionItem(-35.36235859999997, 149.1645425400000002));

        Mission.MissionPlan missionPlan = new Mission.MissionPlan(missionItems);
        Log.d(TAG,"About to upload " + missionPlan.getMissionItems().size() + " mission items");

        Log.d(TAG,"Uploading mission...");
        mDrone.getMission().clearMission()
                        .doOnComplete(() -> Log.d(TAG,"clearMission"))
                        .doOnError(throwable -> Log.d(TAG,"Failed to clearMission"))
                .andThen(mDrone.getMission().setReturnToLaunchAfterMission(isReturningToLaunchAfterMission)
                        .doOnComplete(() -> Log.d(TAG,"Upload setReturnToLaunchAfterMission"))
                        .doOnError(throwable -> Log.d(TAG,"Failed to setReturnToLaunchAfterMission")))
                .andThen(mDrone.getMission().uploadMission(missionPlan)
                        .doOnComplete(() -> Log.d(TAG,"Upload succeeded"))
                        .doOnError(throwable -> Log.d(TAG,"Failed to upload the mission " + throwable.getMessage())))
                .subscribe(() -> Log.d(TAG, "UPLOAD ENDED!"), throwable -> { });

    private static Mission.MissionItem generateMissionItem(double latitudeDeg, double longitudeDeg) {
        return new Mission.MissionItem(
                latitudeDeg,
                longitudeDeg,
                100.0f,
                0.5f,
                true,
                0.0f,
                0.0f,
                Mission.MissionItem.CameraAction.NONE,
                0.0f,
                1.0,
                1.0f,
                0.0f,
                0.3f);
    }

Do you have any idea to fix that please?

JonasVautherin, 2023-06-19:
What autopilot is that? Ardupilot?

TeoVal138, 2023-06-19:
Hi @JonasVautherin  
yes, ardupilot

TeoVal138, 2023-06-20:
Do you have any example of uploading mission using MissionRaw plugin please? 
because it still give me the same problem : (
This is my upgraded code, the MissionItem params should be right based on the MISSION_ITEM_INT command description.

        List<MissionRaw.MissionItem> missionItems = new ArrayList<>();
        missionItems.add(generateMissionItem(0,-35.36235859999997, 149.1642425400000002));
        missionItems.add(generateMissionItem(1,-35.36235859999997, 149.1643425400000002));
        missionItems.add(generateMissionItem(2,-35.36235859999997, 149.1644425400000002));
        missionItems.add(generateMissionItem(3,-35.36235859999997, 149.1645425400000002));

        Log.d(TAG,"Uploading mission...");
        mDrone.getMissionRaw().clearMission()
                        .doOnComplete(() -> Log.d(TAG,"clearMission"))
                        .doOnError(throwable -> Log.d(TAG,"Failed to clearMission"))
                .andThen(mDrone.getMissionRaw().uploadMission(missionItems)
                        .doOnComplete(() -> Log.d(TAG,"Upload succeeded"))
                        .doOnError(throwable -> Log.d(TAG,"Failed to upload the mission " + throwable.getMessage())))
                .subscribe(() -> Log.d(TAG, "UPLOAD ENDED!"), throwable -> { });

private static MissionRaw.MissionItem generateMissionItem(int seq, double latitudeDeg, double longitudeDeg) {
        return new MissionRaw.MissionItem(
                seq, 6, 73, 0, 1, 0f, 0f, 0f, 0f,
                (int) latitudeDeg * 10000000, (int) longitudeDeg * 10000000, 50f, 0);
    }

<html>
<body>

Field Name | Type | Values | Description
-- | -- | -- | --
target_system | uint8_t |   | System ID
target_component | uint8_t |   | Component ID
seq | uint16_t |   | Waypoint ID (sequence number). Starts at zero. Increases monotonically for each waypoint, no gaps in the sequence (0,1,2,3,4).
frame | uint8_t | MAV_FRAME | The coordinate system of the waypoint.
command | uint16_t | MAV_CMD | The scheduled action for the waypoint.
current | uint8_t |   | false:0, true:1
autocontinue | uint8_t |   | Autocontinue to next waypoint. 0: false, 1: true. Set false to pause mission after the item completes.
param1 | float |   | PARAM1, see MAV_CMD enum
param2 | float |   | PARAM2, see MAV_CMD enum
param3 | float |   | PARAM3, see MAV_CMD enum
param4 | float |   | PARAM4, see MAV_CMD enum
x | int32_t |   | PARAM5 / local: x position in meters * 1e4, global: latitude in degrees * 10^7
y | int32_t |   | PARAM6 / y position: local: x position in meters * 1e4, global: longitude in degrees *10^7
z | float |   | PARAM7 / z position: global: altitude in meters (relative or absolute, depending on frame.
mission_type ** | uint8_t | MAV_MISSION_TYPE | Mission type.

</body>
</html>

JonasVautherin, 2023-06-21:
There is a discussion about ardupilot [here](https://github.com/mavlink/MAVSDK/issues/2082#issuecomment-1600597123), where you could maybe connect with some people about the mission stuff.