mavlink/MAVSDK-Java #133: Call System.getParam(). getAllParams() will crash when connected to the drone

GanAHE, 2023-06-23:
**1.description**
I'm trying to run the sample code on Android Studio, and it works fine on version 1.1.1. But it crashes when I try to upgrade to the latest version of 1.3.x. 

**2. code**
private void runMavsdkServer() {
    MavsdkEventQueue.executor().execute(() -> {
      int mavsdkServerPort = mavsdkServer.run();
      drone = new System(BACKEND_IP_ADDRESS, mavsdkServerPort);

      disposables.add(drone.getTelemetry().getFlightMode().distinctUntilChanged()
              .subscribe(flightMode -> logger.debug("flight mode: " + flightMode)));
      disposables.add(drone.getTelemetry().getArmed().distinctUntilChanged()
              .subscribe(armed -> logger.debug("armed: " + armed)));
      disposables.add(drone.getTelemetry().getPosition().subscribe(position -> {
        LatLng latLng = new LatLng(position.getLatitudeDeg(), position.getLongitudeDeg());
        viewModel.currentPositionLiveData.postValue(latLng);
      }));

      isMavsdkServerRunning = true;
      runOnUiThread(() -> buttonRunDestroyMavsdkServer.setText(R.string.destroy_mavsdk_server));

      //================================= Start ==================================
      drone.getParam().getAllParams().doOnSuccess(new Consumer<Param.AllParams>() {
        @Override
        public void accept(Param.AllParams allParams) throws Exception {
          Log.d("MavSDK", "accept: ");
        }
      }).doOnError(new Consumer<Throwable>() {
        @Override
        public void accept(Throwable throwable) throws Exception {

        }
      }).subscribe();

    });

    //================================= End ==================================
  }

**3. version**
    （1）mavsdk:1.2.0 ， mavsdk-server:1.2.1
    （2）mavsdk:1.3.0 ， mavsdk-server:1.3.0
I wanted to try the latest 1.3.2 release, but the IDE reported an error that the version  of MavsdkServer class  didn't match. I didn't know it was fixed or not in the latest version.

**4. error info**
A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x78 in tid 5909 (mavsdk-event-qu), pid 5581 (k.androidclient)

JonasVautherin, 2023-06-23:
Hmm there is apparently a problem in the C++ build. I'll have a look ASAP, probably sometime next week

GanAHE, 2023-06-23:
Amazing work efficiency! Below is the complete console stack information. You can easily reproduce this problem in the MavSDK-Smaple code. I don't think it has anything to do with issue  #131  。

================== stack ======================
I/Mavsdk: MAVSDK version: v1.4.10
D/MAVSDK-Server: Running mavsdk_server with connection url: udp://:14540
I/Mavsdk: Waiting to discover system on udp://:14540...
I/Mavsdk: New system on: 192.168.0.104:54834 (with sysid: 1)
D/Mavsdk: New: System ID: 1 Comp ID: 1
D/Mavsdk: Component Autopilot (1) added.
W/Mavsdk: Vehicle type changed (new type: 2, old type: 0)
D/Mavsdk: Discovered 1 component(s)
I/Mavsdk: System discovered
W/mavsdk-event-qu: type=1400 audit(0.0:6701807): avc: denied { read } for pid=14367 name="somaxconn" dev="proc" ino=177036984 scontext=u:r:untrusted_app:s0:c40,c259,c512,c768 tcontext=u:object_r:proc_net:s0 tclass=file permissive=0
I/Mavsdk: Server started
I/Mavsdk: Server set to listen on 0.0.0.0:33171
D/MAVSDK-Server: mavsdk_server is now running, listening on port 33171
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getLong(Ljava/lang/Object;J)J (greylist,core-platform-api, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->objectFieldOffset(Ljava/lang/reflect/Field;)J (greylist,core-platform-api, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->putLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getLong(Ljava/lang/Object;J)J (greylist,core-platform-api, linking, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->peekLong(JZ)J (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->pokeLong(JJZ)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->pokeInt(JIZ)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->peekInt(JZ)I (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->pokeByte(JB)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->peekByte(J)B (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->pokeByteArray(J[BII)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Llibcore/io/Memory;->peekByteArray(J[BII)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->arrayBaseOffset(Ljava/lang/Class;)I (greylist,core-platform-api, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->arrayIndexScale(Ljava/lang/Class;)I (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getLong(Ljava/lang/Object;J)J (greylist,core-platform-api, reflection, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->putLong(Ljava/lang/Object;JJ)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, reflection, allowed)
W/k.androidclien: Accessing hidden field Ljava/nio/Buffer;->address:J (greylist, reflection, allowed)
V/Telemetry: [mavsdk-event-queue] - Building channel to 127.0.0.1:33171
V/Param: [mavsdk-event-queue] - Building channel to 127.0.0.1:33171
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
D/MavSDK: accept: 
W/k.androidclien: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x78 in tid 14485 (mavsdk-event-qu), pid 14367 (k.androidclient)
E/RtgSchedManager: endActivityTransaction: margin state not match

akirahrkw, 2023-08-01:
Hi, @GanAHE   May I know if you have solved this issue, as I also recently got same crash and error.

version - mavsdk:1.3.1 , mavsdk-server:1.3.2

GanAHE, 2023-09-03:
> Hi, @GanAHE May I know if you have solved this issue, as I also recently got same crash and error.
> 
> version - mavsdk:1.3.1 , mavsdk-server:1.3.2

Hmm, not yet

closed 2024-11-23 by GanAHE